schreibfaul1/ESP32-MiniWebRadio #530: Bluetooth problems

valera39, 2024-10-22:
Hello Dear Wolle!
I still continue to have problems identifying the bluetooth module. BT-Emitter: KCX_BT_Emitter not found. Accordingly, the buttons on the display and on the page do not work. I replaced the module with another one from radio version 3.4 where everything works fine, but in this version it is also not detected.
And what is interesting, if I turn on the bluetooth headphones, they connect to the radio and there is a broadcast of sound from the radio. But all the buttons are still not active.
What do you think?

schreibfaul1, 2024-10-22:
Hi valera,
this is probably the same problem as #524. I have no idea what is causing this. The KCX_BT_Emitter has its own library. I have now added an internal pull-up
[image]
_KCX_BT_Emitter.cpp_
It is conceivable that UART will behave differently in the new IDF V5.
Alternatively, could a 10KOhm resistor or greater on the TX and RX pin help?

valera39, 2024-10-22:
According to the schematic, I have 4.7k resistors. But I also tried without them, but still the module is not visible

schreibfaul1, 2024-10-23:
Yes, that was rubbish with the resistors.
But I still have no idea what the cause is. It's like searching in the fog.
I now clear the buffer of the serial interface after the start to remove any existing characters.
Hope it helps.
If not, please enter _btp<enter>_ in the terminal. This will generate a protocol of the serial interface. It should look like this. Maybe it's different for you?
[image]

It is important that the first command is "AT+" and the response is "OK+".
Thank you for your help.

valera39, 2024-10-23:
I applied your latest updates and the bluetooth module is detected. I don't quite understand how to enter commands in the terminal, please explain the sequence of my actions in detail.

I open a terminal window, write "btp" and the response is that the command does not exist. Or is it necessary to write the command in the serial monitor window? 

I'll continue testing

ve2cfb, 2024-10-23:
Hello Wolle and Valera,

For a while, i thought i was going crazy.

Just a quick update...
By the way, My Shematic shows two 47K instead of a 4.7k
Presently my board has those two resistors installed.
i downloaded Version 3.5k - Oct 23/2024
did a FULL CLEAN, ERASE FLASH and UPLOAD
in KCX_BT_Emitter.cpp around line 52 i dont see a INPUT_PULLUP.  (Probably my eyesight)
Once compiled, the BT button is active (so far so good)
Power off / Power on the unit, BT button disabled. 
(Similar problem)

Thanks Wolle,  i wish i could help a lot more.

valera39, 2024-10-23:
[ve2cfb](https://github.com/ve2cfb),I made a mistake, just put the comma wrong :), of course 47 kOhm.

schreibfaul1, 2024-10-23:
@valera39 
There is no input field in the terminal. You simply click with the mouse on the serial terminal and then type the letters btp for "blue tooth protocol" You cannot see the input, but the terminal responds when you press enter.

ve2cfb, 2024-10-31:
Hello Wolle,
Just wandering...
Is the BT problem fixed...? did i missed something ?

Thanks again.

schreibfaul1, 2024-10-31:
There have been some changes in the KCX_BT_Emitter library to make the UART interface more stable. In addition, the receive buffer is cleared before the first commands are sent.
This entry should appear in the log a few seconds after the start:
[image]
If this does not happen, you could reset the BT emitter by pressing and holding the button. Perhaps the baud rate has changed and is preventing communication.

valera39, 2024-10-31:
I have version 1.7 written on the module, but in the terminal the module is defined as V 1.4. Is this normal?

schreibfaul1, 2024-10-31:
You must mean this:
[image]
I have two boards. V1.7 is printed on them and 1.4 is read from the chip.
Is the BT_Emitter running reliably or are you having similar problems to ve2cfb?

valera39, 2024-10-31:
I too have two boards version 1.7 and both are detected as 1.4. Another module should be arriving soon, I'll see how it is detected.
I have one module in radio version 3.4 and one in radio version 3.5. The one in the receiver 3.4 subjectively evaluating works more stable.

ve2cfb, 2024-11-01:
Hello Wolle,
Just installed Version 3.5q - Nov 01/2024 and the BT Button problem seems to be working so far.
i did try a couple of combinations and everything works ok.
Have you change something...?
Will do more test.

But... . i cannot figure out the following...
Using the web application, when trying to add stations from Germany, the stations list is still empty.
I did check other Country and no problem there... Go figure.

Have a nice day.

schreibfaul1, 2024-11-02:
That is strange. The function only runs in the browser using JS. I simply query the database.
You can also do this by typing https://de1.api.radio-browser.info/json/countries#name into your browser.
I have now changed the sorting criterion. First comes the country code, then the country name.

[image]

Maybe that will solve your problem with "DE"

ve2cfb, 2024-11-02:
Hello Wolle,
It is a lot better. Now i am able to get Germany.  it worked 20 times and failed 2 times. 
Many thanks...

Concerning the BT button problem, here is the message from the BTP command when problem exist..

---- Opened the serial port COM15 ----
---- Sent utf8 encoded message: "btp" ----
15:29:56 Terminal  :  btp
[604490][E][main.cpp:2771] loop(): TX: AT+
[604498][E][main.cpp:2771] loop(): TX: AT+ADDLINKADD=263510091300
[604505][E][main.cpp:2771] loop(): TX: AT+VMLINK?
[604510][E][main.cpp:2771] loop(): TX: AT+VMLINK?
[604516][E][main.cpp:2771] loop(): TX: AT+RESET
[604521][E][main.cpp:2771] loop(): TX: AT+RESET
15:30:10 click on ..  backpane
15:30:11 click on ..  backpane

NOTE:  One way to get the BT functional again, is delete de settings.json (using the PC) Power on the unit again and you are in business.

schreibfaul1, 2024-11-02:
Thank you for the protocol. It can be seen that the BT emitter is not responding. 
The UART interface has been changed with the new IDF version V5. I will stay on the subject.

valera39, 2024-11-03:
Confirmed, the bluetooth module is not detected

schreibfaul1, 2024-11-03:
I have noticed that the UART interface (Serial2) is not operated as long as a connection to a radio station is being established. It takes even longer for encrypted connections. If the BT emitter responds during this time, the information is lost. I have no idea why this is the case. Now the BT emitter is requested afterwards and that works better.

valera39, 2024-11-07:
It looks like you've defeated the bluetooth module detection issue, but the module's current state settings are not saved after a reboot

schreibfaul1, 2024-11-11:
Right, I don't have the BT mode (RX/TX) in the settings.json yet. I'll catch up on that soon.

closed 2024-12-27 by github-actions[bot]