schreibfaul1/ESP32-MiniWebRadio #531: Afterimage

samand587, 2024-10-23:
When switching from radio mode to clock mode, an afterimage remains. How can I get rid of it?
[image]

samand587, 2024-10-29:
Dear author! Is my question so stupid that there is no answer to it?

schreibfaul1, 2024-10-29:
Hello samand587,
I see your screenshot but don't know the cause. This is probably the first time you have encountered this error. It is possible that the variable _clockSubMenue is not equal to 0 when the CLOCK mode is called. I have added this and hope that the error has been eliminated. Please update the main.cpp.

samand587, 2024-10-30:
Hello, the author! Uploaded a sketch with a correction. The afterimage has disappeared. But the time zone setting has disappeared in the browser. How can I fix this?

schreibfaul1, 2024-10-30:
Thanks for the feedback.
I think that the missing time zone data has another cause. Perhaps the timezones.csv file is corrupt or damaged. You can find this file in the root of the SD card.

samand587, 2024-10-31:
Hi! I replaced the file - everything is fine!

closed 2024-10-31 by samand587