schreibfaul1/ESP32-MiniWebRadio #533: radio freeze

valera39, 2024-10-24:
Hello Dear Wolle!
I noticed another bug, if you go to a station, stream that is currently not available or to the wrong link, the radio hangs and does not respond to anything. After a while the radio comes back to life and switches to the previous station that was playing, and the screen remains the logo and the name of the station to which there is no connection, but the tags are displayed station that was playing before that 
Here is a bit of the monitor log:
23:09:24 action: ...  switch to station 67
23:09:24 AUDIO_info:  buffers freed, free Heap: 132067 bytes
23:09:24 AUDIO_info:  connect to: "cast1.torontocast.com" on port 2130 path "/stream"
[815711][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
23:11:24 AUDIO_info:  Request https://cast1.torontocast.com:2130/stream failed!
[815723][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
23:11:24 AUDIO_info:  Request https://cast1.torontocast.com:2130/stream failed!
23:11:24 StationName:
23:11:24 StreamTitle:
[815750][E][NetworkClient.cpp:429] write(): fail on fd 61, errno: 113, "Software caused connection abort"
[818313][W][STA.cpp:135] _onStaArduinoEvent(): Reason: 201 - NO_AP_FOUND
[818320][W][STA.cpp:537] disconnect(): STA already disconnected.
23:11:30 AUDIO_info:  buffers freed, free Heap: 135383 bytes
23:11:30 AUDIO_info:  connect to: "stream.cassiopeia-station.ru" on port 1045 path "/stream"
23:11:30 AUDIO_info:  SSL has been established in 494 ms, free Heap: 135159 bytes
23:11:30 bitRate:     320Kbit/s
23:11:30 StationName: Всесоюзное радио онлайн
23:11:30 AUDIO_info:  MP3Decoder has been initialized, free Heap: 134195 bytes , free stack 3436 DWORDs
23:11:30 lastURL: ..  https://stream.cassiopeia-station.ru:1045/stream
23:11:30 AUDIO_info:  stream ready
23:11:30 AUDIO_info:  syncword found at pos 0
23:11:30 AUDIO_info:  MPEG-2.5, Layer I
23:11:30 AUDIO_info:  Channels: 2
23:11:30 AUDIO_info:  SampleRate: 44100
23:11:30 AUDIO_info:  BitsPerSample: 16
23:11:30 AUDIO_info:  BitRate: 320000
23:11:30 StreamTitle: Танцуем слушаем поем-1986.19.11
23:11:31 Audiocodec:  MP3
23:11:39 webSrv: ...  Load from SD  "/stations.json"
23:11:39 webSrv: ...  Load from SD  "/ir_buttons.json"
Can this be fixed?
Thank you!

schreibfaul1, 2024-10-25:
This looks like a WiFi problem. 
ASSOC_LEAVE means that the last connected access point is currently busy.
Another AP is searched for in a mesh network.
NO_AP_FOUND means no access point available for the enquiry.

I had activated Arduino version 3.0.7 yesterday. Due to reports of problems with PSRAM, I went back to Arduino 3.0.5. Maybe this will help in your case and V3.0.7 also has a WiFi problem?

valera39, 2024-10-25:
I tried Arguino 3.0.5/IDF 5.1.4 Unfortunately the problem did not disappear. If I switch to a non-working stream, after a while the logo and name of the non-working station appears on the screen, but the sound and song title are played from the previous station. This is not easy to describe accurately with an online translator, but I hope you can understand me.

schreibfaul1, 2024-10-26:
If the desired URL is not reached, it jumps back to the last working URL( _lastconnectedhost).  Does this only happen with https://cast1.torontocast.com:2130/stream? My guess is that it happens with many SSL streams.
What about the non-SSL version? http://cast1.torontocast.com:2130/stream

valera39, 2024-10-26:
The thing is that these links have recently been blocked in my country. And now they are played only through VPN.
It's strange that the logo and name are from one station and the sound and tags are from another station

schreibfaul1, 2024-10-26:
This is a mistake, I want the correct logo to be displayed. I'll have a look at that soon.

valera39, 2024-10-26:
I should add that this station was chosen simply as an example for what happens on the display when switching from an available station to an unavailable station.

valera39, 2024-10-26:
While I was writing, you already replied:)
Thanks!

valera39, 2024-11-03:
Hello Dear Wolle!
There's that nasty problem again with the radio freezing on an idle station in the list.
When switching to a station that currently has no stream, the radio freezes, the screen freezes, the time does not update, the touchscreen and remote do not respond. After about two minutes the time starts to update, the screen shows the logo of the previous station without the name of the artist and the name of the song and silence.
Is it possible to fix this annoying problem, so that the radio, say, when the station is not working at the moment in the list (and this sometimes happens with many stations), automatically switched to the next station in the list and would start playing it
Part of the error log:
18:35:33 AUDIO_info:  connect to: "evcast.mediacp.eu" on port 2075 path "/stream"
[183465][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
18:37:33 AUDIO_info:  Request https://evcast.mediacp.eu:2075/stream failed!
18:37:33 AUDIO_info:  Request https://evcast.mediacp.eu:2075/stream failed!
18:37:33 StationName:
18:37:33 StreamTitle:
A two-minute wait seems like a long time to me.
I look forward to your reaction to my message,thank you!

valera39, 2024-11-05:
Is there a solution to my question?

schreibfaul1, 2024-11-06:
Hi valera39,
This is a difficult issue. Let's assume that https://evcast.mediacp.eu:2075/stream works. And you would enter a URL that does not exist, e.g. ...2076/stream then it automatically switches back to 2075/stream.
[image]
However, this is completely different if 2075/stream no longer works, because 2075/stream is then called up endlessly.

I have now installed a counter that prevents this, then a voice message appears and the endless loop is cancelled.
[image]
This is difficult to test, you need a URL that works and later stops working.

valera39, 2024-11-07:
Hello Dear Wolle!
Can you make it possible to switch from the idle stream to the next station in the list, not the previous one. Agree, it would be more convenient for the user. If you press forward on the remote again, the selection will be the idle station, then it will roll back to the previous working station and so on in a circle. Consequently, it will be necessary to manually enter the number of the working station, which is inconvenient, as users most often use the forward and back buttons to switch between stations.
I hope you understand the logic of my explanation.

schreibfaul1, 2024-11-08:
That's pretty inconvenient. I have removed all this, but now there is a new error message in the display response from the server (this is not new)
[image]
Server is not responding (is new)
[image]

closed 2024-12-24 by github-actions[bot]