schreibfaul1/ESP32-MiniWebRadio #536: Compilation error ESP32S3.

hevet, 2024-10-29:
An error appears at the end of compilation. Tested on a laptop and computer, under the logo

c:/users/jacek/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32s3_OTA/src/main.cpp.o: in function `find_first_of(char const*, char const*, int)':
C:\Users\Jacek\Documents\PlatformIO\Projects\ESP32-MiniWebRadio-master/src/common.h:445: multiple definition of `find_first_of(char const*, char const*, int)'; .pio/build/esp32s3_OTA/src/audiocontroltask.cpp.o:C:\Users\Jacek\Documents\PlatformIO\Projects\ESP32-MiniWebRadio-master/src/common.h:445: first defined here
c:/users/jacek/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32s3_OTA/src/main.cpp.o:C:\Users\Jacek\Documents\PlatformIO\Projects\ESP32-MiniWebRadio-master/src/main.cpp:206: multiple definition of `audio'; .pio/build/esp32s3_OTA/src/audiocontroltask.cpp.o:C:\Users\Jacek\Documents\PlatformIO\Projects\ESP32-MiniWebRadio-master/src/audiocontroltask.cpp:8: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32s3_OTA\firmware.elf] Error 1

schreibfaul1, 2024-10-29:
That is strange. The audiocontroltask no longer exists. I moved the task to the audioI2S library some time ago. But just a few days ago I deleted audiocontroltsak.cpp from the repository. But somehow the file still exists for you. I have no idea why this is the case. Please check if the file in the src folder still exists and delete the build folder before compiling.
[image]

hevet, 2024-10-29:
@schreibfaul1 Thanks. Deleting the file helped.

closed 2024-10-29 by hevet