schreibfaul1/ESP32-MiniWebRadio #537: Download error

valera39, 2024-10-29:
Hello Dear Wolle!
Updated to the latest version. The initial loading is normal. If the internet is down or the connection is unstable, if you restart the radio, it does not load anymore. 

Here is the log: 
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.5
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.5o - Oct 29/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 6015520 bytes
RESET_REASON: Software reset due to exception/panic

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 29789.5 MB by 29818.0 MB free
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556d5  PS      : 0x00060e30  A0      : 0x8211c042  A1      : 0x3fccbb80
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fccbb40
A10     : 0x3fccbf89  A11     : 0x3c141273  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff

Backtrace: 0x400556d2:0x3fccbb80 0x4211c03f:0x3fccbb90 0x421199c9:0x3fccbeb0 0x42009e83:0x3fccbf70 0x42012b96:0x3fccbff0 0x4201af81:0x3fccc060 0x4208cf40:0x3fccc0f0
  #0  0x4211c03f in _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1524
  #1  0x421199c9 in sprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/sprintf.c:618
  #2  0x42009e83 in updateSettings() at src/main.cpp:557
  #3  0x42012b96 in defaultsettings() at src/main.cpp:447
  #4  0x4201af81 in setup() at src/main.cpp:1441 (discriminator 7)
  #5  0x4208cf40 in loopTask(void*) at C:/Users/admin/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59

ELF file SHA256: c05c2e4d

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40391739
  #0  0x40391739 in xthal_save_extra_nw at ??:?

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1160
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa10
load:0x403cc700,len:0x2bf0
entry 0x403c988c
On the SD card, after rebooting, the file names look like this
[image]
I hope you get this fixed soon.
The new format of settings and radio stations is very good!
Hoping for a fix soon

schreibfaul1, 2024-10-29:
Thanks for the error report. I only changed this yesterday. I formatted the json files more nicely and took the opportunity to add the value "lastconnctedfile" to the settings.json. This is for later. With the error "lastconnectedfile" is NULL and this causes _Core 1 panic'ed (LoadProhibited). Exception was unhandled_
As soon as an audio file is played, the value is set. An empty string is now generated for the initial initialisation. This should solve the problem.

valera39, 2024-10-29:
Thanks for the fix and detailed reply, it's now as it should be. Just to clarify, I have two access points, both of which are defined in the networks file. I forcibly disconnected one access point, but the radio did not connect to the other one, the backup one. Is this how it should be?

valera39, 2024-10-30:
Would you be so kind as to answer the question if possible?

schreibfaul1, 2024-10-31:
My code for switching to another AP is not effective. And the WiFi class doesn't seem to do this automatically. My idea is that if an AP fails or is no longer available, the system searches for another available AP. Regardless of whether this is in your own network or in another (registered) network. I am still testing this.

valera39, 2024-10-31:
I hear you, thank you!
Good luck with your work

closed 2024-10-31 by valera39