schreibfaul1/ESP32-MiniWebRadio #538: No image / no sound

Valentin211, 2024-10-30:
Hello Wolle,

I tested ESP32-MiniWebRadio with the following components: ESP32-S3 N8R8, ILI9341-2.8", Max98357.
The compilation and loading in VS Code was done without errors, but the radio does not work properly:
- Vcc display and Vcc I2S audio powered at +5V ESP32 module - there is sound, the radio station is running but there is no image on the display (Backlight turns on for a few seconds and then remains off);
- Vcc display and Vcc I2S audio powered at +3.3V ESP32 module - there is an image on the display, VU Meter is active, Touchscreen seems OK (I did not insist in details) but there is no sound.

Thank you in advance.

schreibfaul1, 2024-10-30:
It is possible that the 5V pin of the board has no voltage. I think a jumper needs to be soldered in first.

skassel2024, 2024-10-30:
A jumper has to be soldered. Also check the diodes next to the jumper. I had the same problem at first. However, one of the three diodes was defective in my case

[image]

Valentin211, 2024-10-30:
> A jumper has to be soldered.  Thank you skassel2024 , now it's working.

[url=https://postimages.org/][img]https://i.postimg.cc/3NDvyRXX/IMG-20241030-192957697.jpg[/img][/url]

Thank you Wolle, it's great project!

closed 2024-10-30 by Valentin211