schreibfaul1/ESP32-MiniWebRadio #539: Weboberfläche Fehlerausgabe im Terminal

skassel2024, 2024-10-30:
Hallo Wolle,
beim Aufruf der Weboberfläche Menü Einstellunge/IR Settings erscheint in der Terminalausgabe folgende Fehlermeldung.
Die Datei "Remote_Control_Yellow.png" ist aber auf der SD vorhanden
Gruß Sören
[image]

schreibfaul1, 2024-10-30:
Das ist in der Tat seltsam. Der Dateiname ist richtig geschrieben, aber die SD Karte meinte dass die Datei nicht vorhanden ist.
Alle Grafiken werden beim ersten Mal in den Cache des Browsers geladen und bleiben dort die maximale Zeit (90 Tage?) erhalten. Das mache ich mittels einem versteckten html-div.
So wird vermieden, dass die Grafiken z.B. beim Drücken eines Buttons nachgeladen werden.
Wenn du auf Entwicklertools gehst und dann "Cache leeren und vollständig aktualisieren" klickst, kommt der Fehler dann wieder?

skassel2024, 2024-10-31:
Hallo Wolle,
ja exakt. Der Fehler erscheint nach dem Löschen des Browsercache wieder. Also scheint beim ersten Laden der Symbole etwas schief zu laufen. Nach dem zweiten Aufruf des Menues "IR Settings" tritt die Fehlermeldung  nicht mehr auf.

schreibfaul1, 2024-10-31:
Vielleicht ist im gesamten Prozess irgendwas nicht optimal.
Die Buttons auf der Webseite sind img Objekte. Beim Mausklick wird eine Datei angefordert.
In index.h ist das in dem Fall
[image]
und das kommt in main.cpp so an:
[image]
Und bis dahin scheint es okay zu sein
scaleImage() macht mit png Dateien nichts  (ist für jpg bei unterschiedlicher Displayauflösung)
streamfile() öffnet die Datei (wahrscheinlich ist hier die Ursache) und sendet sie an die Webseite.

closed 2024-11-06 by skassel2024