schreibfaul1/ESP32-MiniWebRadio #540: Issue with ESP32-S3-DevKitC-1-N8R2 board

shsksa, 2024-10-30:
First of all, I would like to thnak the author for the great work. 

Secondly, When choosing this board (ESP32-S3-DevKitC-1-N8R2) and configure PSRAM through memconfig , the project compiles and uploads to the board. However, the board does not boot and the following is the terminal message, what am I doing wrong?

Thanks in advance. 

=========  Terminal messages  =====================

ELF file SHA256: 3b2c5192

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037823c
  #0  0x4037823c in esp_restart_noos at C:/Users/SHS/.platformio/packages/framework-espidf/components/esp_system/port/soc/esp32s3/system_internal.c:158 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1160
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa10
load:0x403cc700,len:0x2c04
entry 0x403c988c
E (698) mmu_psram: SPI RAM space not enough for the instructions, need to copy to 2359296 B.
E (698) esp_psram: No enough psram memory for rodata!

abort() was called at PC 0x420a64ff on core 0

Backtrace: 0x40378652:0x3fceb200 0x4037f8d5:0x3fceb220 0x4038c352:0x3fceb240 0x420a64ff:0x3fceb2b0 0x40377c2a:0x3fceb2f0 0x403cd94e:0x3fceb340 0x403cdc3d:0x3fceb380 0x403c98f5:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
  #0  0x40378652 in panic_abort at C:/Users/SHS/.platformio/packages/framework-espidf/components/esp_system/panic.c:466
  #1  0x4037f8d5 in esp_system_abort at C:/Users/SHS/.platformio/packages/framework-espidf/components/esp_system/port/esp_system_chip.c:84
  #2  0x4038c352 in abort at C:\Users\SHS\.platformio\packages\framework-espidf\components\newlib/abort.c:38
  #3  0x420a64ff in esp_psram_init at C:\Users\SHS\.platformio\packages\framework-espidf\components\esp_psram/esp_psram.c:170 (discriminator 3)
  #4  0x40377c2a in call_start_cpu0 at C:/Users/SHS/.platformio/packages/framework-espidf/components/esp_system/port/cpu_start.c:481 (discriminator 3)

schreibfaul1, 2024-10-30:
Your board has 2MB quad PSRAM. In sdkconfig, octal PSRAM is set by default. 
This is what you do:
[image]

1) Check that ESP32S3 is selected
2) Click on the ant
3) Select Run Menuconfig
4) Component config --->

**ESP PSRAM --->** select
**SPI RAM config -->** select
**Mode (QUAD/OCT) of SPI RAM chip in use (Octal Mode PSRAM)  --->** select
choose Quad Mode PSRAM
esc, esc ... Y

shsksa, 2024-10-31:
Thank you very much for the reply. However, I already done that, otherwise it complains about octal PSRAM initialization. 

It seems that the boot-loader is trying to copy more than 2M to PSRAM (256 byte more than the 2M byte) . Please correct me if I was wrong. 

E (698) mmu_psram: SPI RAM space not enough for the instructions, need to copy to 2359296 B.
E (698) esp_psram: No enough psram memory for rodata!

Is there any workaround? 

Thanks again

shsksa, 2024-10-31:
Thanks a lot. Deactivating the mentioned options did the trick. Now I need to check the performance and update you. 

On the other hand , I was also looking for a solution and I was able to compile and solve the PSRAM size issue with the following environment (so now we have 2 solutions for anyone who might be at my position)

; Flash: 8MB QD, PSRAM: 2MB QD
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

board_build.arduino.memory_type = qio_qspi
board_build.flash_mode = qio
board_build.psram_type = qio
board_upload.flash_size = 8MB
board_upload.maximum_size = 8388608
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM

schreibfaul1, 2024-10-31:
Thanks, I didn't know that. Then you can probably overwrite the settings in menuconfig with **_board_build.psram_type = qio_**.

shsksa, 2024-11-03:
As promised to update you regarding the performance.  So far I have good success in performance. I tested with 64, 96, 128, and 192 kbps. I noticed at higher bit-rate, there are some packet drops especially when accessing the webpage.  But nothing to be considered as a deal breaker.

schreibfaul1, 2024-11-03:
These drop outs cannot be completely avoided, especially if the ping times are too long or TCP packets are lost. It is better to use Arduino as a component than the precompiled version.
`framework = arduino, espidf`
The WiFi and LWIP settings in sdkconfig improve the data throughput.

closed 2024-12-19 by github-actions[bot]