schreibfaul1/ESP32-MiniWebRadio #541: Unreadable SD card

valera39, 2024-10-31:
Hello Dear Wolle!
I had the same error again as I wrote earlier. Suddenly the radio froze and after a forced reboot the memory card suddenly stopped reading, the files on it became like this
[image]

valera39, 2024-10-31:
I formatted the card again and copied the files from the archive and the radio booted again.
Why is this happening?

valera39, 2024-10-31:
And here's the same problem again:
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.5
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.5p - Oct 29/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 6015520 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 29789.5 MB by 29818.0 MB free
E (319) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (320) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (322) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (329) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (335) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (342) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (348) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (355) diskio_sdmmc: sdmmc_write_blocks failed (265)
[  1425][E][vfs_api.cpp:99] open(): /sdcard/settings.json does not exist, no permits for creation
[  1445][E][main.cpp:460] operator()(): index "volume": not found
[  1452][E][main.cpp:460] operator()(): index "volumeSteps": not found
[  1459][E][main.cpp:460] operator()(): index "ringVolume": not found
[  1466][E][main.cpp:460] operator()(): index "volumeAfterAlarm": not found        
[  1474][E][main.cpp:460] operator()(): index "BTvolume": not found
[  1481][E][main.cpp:460] operator()(): index "BTpower": not found
[  1488][E][main.cpp:460] operator()(): index "alarmtime_sun": not found
[  1495][E][main.cpp:460] operator()(): index "alarmtime_mon": not found
[  1503][E][main.cpp:460] operator()(): index "alarmtime_tue": not found
[  1510][E][main.cpp:460] operator()(): index "alarmtime_wed": not found
[  1518][E][main.cpp:460] operator()(): index "alarmtime_thu": not found
[  1525][E][main.cpp:460] operator()(): index "alarmtime_fri": not found
[  1533][E][main.cpp:460] operator()(): index "alarmtime_sat": not found
[  1540][E][main.cpp:460] operator()(): index "alarm_weekdays": not found
[  1548][E][main.cpp:460] operator()(): index "timeAnnouncing": not found
[  1555][E][main.cpp:460] operator()(): index "mute": not found
[  1562][E][main.cpp:460] operator()(): index "brightness": not found
[  1569][E][main.cpp:460] operator()(): index "sleeptime": not found
[  1576][E][main.cpp:460] operator()(): index "station": not found
[  1583][E][main.cpp:460] operator()(): index "toneLP": not found
[  1590][E][main.cpp:460] operator()(): index "toneBP": not found
[  1596][E][main.cpp:460] operator()(): index "toneHP": not found
[  1603][E][main.cpp:460] operator()(): index "balance": not found
[  1610][E][main.cpp:460] operator()(): index "timeFormat": not found
[  1617][E][main.cpp:460] operator()(): index "Timezone_Name": not found
[  1625][E][main.cpp:460] operator()(): index "Timezone_String": not found
[  1632][E][main.cpp:460] operator()(): index "lastconnectedhost": not found       
[  1640][E][main.cpp:460] operator()(): index "lastconnectedfile": not found
[  1648][E][main.cpp:460] operator()(): index "sleepMode": not found
[  1655][E][main.cpp:460] operator()(): index "cur_AudioFileNr": not found
[  1663][E][main.cpp:460] operator()(): index "cur_AudioFolder": not found
[  1670][E][main.cpp:460] operator()(): index "state": not found
E (614) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (614) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (616) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (624) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (629) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (637) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (643) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
[... 8 lines cut ...]
E (704) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (711) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (717) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (724) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (730) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (737) diskio_sdmmc: sdmmc_write_blocks failed (265)
00:00:00 AUDIO_info:  file "/common/m/MiniWebRadioV3.jpg" not found
E (754) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (757) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (763) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (771) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (777) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (784) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (790) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (797) diskio_sdmmc: sdmmc_write_blocks failed (265)
[  1866][E][vfs_api.cpp:310] VFSFileImpl(): fopen(/sdcard/settings.json) failed    
[  1885][E][main.cpp:575] updateSettings(): file "settings.json" not found
00:00:00 setup: ....  seek for WiFi networks
E (830) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (830) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (836) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (843) diskio_sdmmc: sdmmc_write_blocks failed (265)
[  1913][E][vfs_api.cpp:99] open(): /sdcard/networks.csv does not exist, no permits for creation
[  4756][E][WiFiMulti.cpp:306] run(): [WIFI] no matching wifi found!
00:00:00 WiFI_info:   WiFi credentials are not correct

valera39, 2024-10-31:
Changing the card didn't help

valera39, 2024-10-31:
I noticed that problems occur when actively using the touchscreen or the remote control buttons

schreibfaul1, 2024-11-01:
Ooh, read error from the SD card. This usually has a mechanical cause. Perhaps it is a contact problem or a bad solder joint.

skassel2024, 2024-11-01:
Hello valera39,
 I also had the problem with the SD card at the beginning. See my answer in
issue #538. At that time, one of the three diodes was defective, so that the voltage to TFT was below 3V. It is also better to bridge the three resistors on the display to the SD socket.

valera39, 2024-11-01:
[skassel2024](https://github.com/skassel2024)
Thanks for the advice, but I think my power supply and contacts are fine. I can not understand why there is periodic overwriting of files in unreadable format (see my first post in the thread). This has never happened before and does not happen now on other radios I have built, but on earlier firmware versions.

schreibfaul1, 2024-11-01:
@valera39 
this is most likely a mechanical problem, sdmmc_send_cmd returned 0x109 means CRC error.
Or your SD card does not manage 80MHz, then you can lower the frequency.

valera39, 2024-11-02:
The card may have started to fail even though it's fairly new. I tried different frequencies and left it at default. I put in another card and so far everything is working
I'll be watching, thanks!

closed 2024-12-18 by github-actions[bot]