schreibfaul1/ESP32-MiniWebRadio #543: Error: Code doesn't compile, KCX Error

HWH53, 2024-11-01:
Hallo Wolle,
ich wollte soeben nach dem Urlaub von der Version v3.05a auf den aktuellen Softwarestand updaten und laufe beim Compilieren in einen Fehler!
Dem Protokoll entnehme ich dass das Problem irgendwo im Bereich des 'class KCX_BT_Emitter' zu liegen scheint. Ich verwende aber keinen Bluetooth Emitter.
Das Installationsprotokoll habe ich beigefügt. Habe ich irgendetwas übersehen?

mfg
Helmut 

`*  Task wird ausgeführt: C:\Users\HWH\.platformio\penv\Scripts\platformio.exe run --environment esp32s3_OTA 

Processing esp32s3_OTA (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/pioarduino/platform-espressif32/releases/download/51.03.05/platform-espressif32.zip; framework: arduino, espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP32-S3-DevKitC-1-N16R8.html
PLATFORM: Espressif 32 (51.3.5) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.5
 - framework-arduinoespressif32-libs @ 5.1.0+sha.33fbade6b8
 - framework-espidf @ 3.50104.0 (5.1.4)
 - tool-cmake @ 3.21.3
 - tool-esptoolpy @ 4.7.5
 - tool-mklittlefs @ 3.2.0
 - tool-ninja @ 1.9.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32-audioI2S @ 2.0.0+sha.9cfea80
|-- ESP32-KCX-BT-EMITTER @ 0.0.0+20240128094011.sha.b766d32
|-- ESP32-DLNA-Client @ 0.0.0+20231205185914.sha.c320f5e
|-- ESP32-IR-Remote-Control @ 1.0.10+sha.51aa0ce
|-- ac101
|-- BH1750
|-- es8388
|-- ftpsrv
|-- rtime
|-- SPIFFS @ 3.0.5
|-- tftLib
|-- websrv
|-- ArduinoOTA @ 3.0.5
|-- ESPmDNS @ 3.0.5
|-- FFat @ 3.0.5
|-- FS @ 3.0.5
|-- Preferences @ 3.0.5
|-- SD_MMC @ 3.0.5
|-- SPI @ 3.0.5
|-- Ticker @ 3.0.5
|-- WiFi @ 3.0.5
|-- NetworkClientSecure @ 3.0.5
|-- Wire @ 3.0.5
Building in release mode
Generating assembly for .pio\build\esp32s3_OTA\https_server.crt.S
Generating assembly for .pio\build\esp32s3_OTA\rmaker_mqtt_server.crt.S
Generating assembly for .pio\build\esp32s3_OTA\rmaker_claim_service_server.crt.S
Generating assembly for .pio\build\esp32s3_OTA\rmaker_ota_server.crt.S
Compiling .pio\build\esp32s3_OTA\src\main.cpp.o
Generating LD script .pio\build\esp32s3_OTA\memory.ld
Compiling .pio\build\esp32s3_OTA\app_trace\app_trace.c.o
Compiling .pio\build\esp32s3_OTA\app_trace\app_trace_util.c.o
[... 154 lines cut ...]
In file included from C:/Users/HWH/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:72,
                 from C:/Users/HWH/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from src/common.h:27,
                 from src/main.cpp:1:
src/main.cpp:2770:40: error: 'class KCX_BT_Emitter' has no member named 'list_protokol'
 2770 |                 log_e("%s", bt_emitter.list_protokol(i));
      |                                        ^~~~~~~~~~~~~
C:/Users/HWH/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:246:76: note: in definition of macro 'log_e'
  246 | #define log_e(format, ...)     log_printf(ARDUHAL_LOG_FORMAT(E, format), ##__VA_ARGS__)
      |                                                                            ^~~~~~~~~~~
Compiling .pio\build\esp32s3_OTA\esp_coex\esp32s3\esp_coex_adapter.c.o
Archiving .pio\build\esp32s3_OTA\esp-idf\esp_app_format\libesp_app_format.a
Archiving .pio\build\esp32s3_OTA\esp-idf\esp-tls\libesp-tls.a
Indexing .pio\build\esp32s3_OTA\esp-idf\esp_app_format\libesp_app_format.a
Indexing .pio\build\esp32s3_OTA\esp-idf\esp-tls\libesp-tls.a
Compiling .pio\build\esp32s3_OTA\esp_common\src\esp_err_to_name.c.o
*** [.pio\build\esp32s3_OTA\src\main.cpp.o] Error 1
=============================================================== [FAILED] Took 44.00 seconds ===============================================================
Environment    Status    Duration
-------------  --------  ------------
esp32s3_OTA    FAILED    00:00:44.003
========================================================== 1 failed, 0 succeeded in 00:00:44.003 ==========================================================
 *  Der Terminalprozess "C:\Users\HWH\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32s3_OTA'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 
`

schreibfaul1, 2024-11-01:
Das Protokoll gibts noch nicht lange. Das habe ich eingebaut, um zu sehen was auf der UART Schnittstelle vom KCX BT Emitter passiert.
Gehe auf die Ameise und dann auf "Full Clean" damit die Bibliothek neu geladen wird.

Du kannst im Terminal btp<ENTER> eintippen, einfach blind ins Terminalfenster, dann kommt das Protokoll und sieht etwa so aus:
[image]

HWH53, 2024-11-02:
Hallo Wolle,
danke für die schnelle Antwort! 
Alles arbeitet nun wieder einwandfrei.😅
mfg
Helmut

closed 2024-11-02 by HWH53