schreibfaul1/ESP32-MiniWebRadio #545: WI-Fi level

valera39, 2024-11-03:
Hello Dear Wolle!
Today I upgraded the radio from the old firmware version 3.0 to the latest one. I was very surprised to find that the radio works without intermittent streaming only when I brought the router about 30 cm closer to the radio. Before the update, the radio worked consistently at a meter or more away from the router. I completely rule out a malfunction of my components. The reception conditions have not changed either.
Is it possible that the decrease in sensitivity is due to the new Arduino IDE platform and wifi libraries?
Is there any way to fix this and what are your thoughts on this. 
It makes no sense to switch to the old version, as the radio has grown a lot in functionality.
I will be glad if other users will also join this thread

schreibfaul1, 2024-11-03:
Hello valera,
the receiving amplifier regulates itself, but the transmission power can be changed. The standard setting is 5dBm. This is enough to bridge a distance of 5 metres and a wall for me, 22dBm is the maximum adjustable value with higher power consumption.
[image]

valera39, 2024-11-03:
You probably have a powerful router. Where are these settings?

schreibfaul1, 2024-11-03:
This is in common.h at the top

valera39, 2024-11-03:
So the new Arduino platform has nothing to do with it? Okay, thank you. I'll try playing with the settings.
Please see my question #533
this problem puzzles me

valera39, 2024-11-03:
You'll make a big deal if you can teach the radio to switch between APs in the list. I admire your hard work and support!

closed 2024-12-19 by github-actions[bot]