schreibfaul1/ESP32-MiniWebRadio #546: WMA-Format

skassel2024, 2024-11-04:
Hallo Wolle,
gibt es, oder wird es eine Möglichkeit geben um im Player .wma abbzuspielen? (ohne erst in mp3 zu konvertieren).
Gruß Sören

schreibfaul1, 2024-11-04:
Hallo Sören,
ich glaube, es gibt aus lizenzrechtlichen Gründen keine Möglichkeit einen wma Dekoder einzubauen.
Gruß
Wolle

skassel2024, 2024-11-04:
Hallo Wolle,
gibts damit keine Möglichkeit?
Gruß Sören
[image]

schreibfaul1, 2024-11-04:
Das ist ein HW Dekoder, in Version 1 und 2 war er hier auch im Einsatz. Den VS1053 wieder hinzuzufügen ist einiges an Arbeit. Als Einstieg könntest du gucken, wie ich das damals gemacht habe.

skassel2024, 2024-11-04:
Mit programmieren ist bei mir nicht so viel. Kann mal versuchen es nachzuvollziehen. Aber ich glaube eher, konvertieren meiner Musikfiles wird die Lösung werden ;-)

closed 2024-12-20 by github-actions[bot]