schreibfaul1/ESP32-MiniWebRadio #548: Web Interface stations list Problem

ve2cfb, 2024-11-05:
Hello Wolle,
This was open in #530

[image]

Still having problem when doing a display for country that have a very big list like Germany, US and a couple more.
**Accessing small listing, there is no problem.**
For example, Germany gave me intermittent problem and was able to access it 10 time and it failed about 2 time.
The US country(6538) entry, is a lot worse, cannot display anything at all.

Did a search on your web page but could not find anything. 
Must be on my side of the fence...

Thanks.

schreibfaul1, 2024-11-06:
It looks as if your PC has a problem with loading the list. The ESP32 has nothing to do with it. It may also be that the browser does not like it. Do you have the option of testing a different browser?

ve2cfb, 2024-11-06:
Yes i did a test it with Microsoft Edge and its working OK.
My Google Chrome was at Version 130.0.6723.92  now is at 130.0.6723.117 and everything works OK.
Thanks for the tip.

closed 2024-12-22 by github-actions[bot]