schreibfaul1/ESP32-MiniWebRadio #551: websrv minimize timeouts

skassel2024, 2024-11-06:
Hallo Wolle,
seit der letzten Änderung werden die Senderlogos nicht mehr aktualisiert. Nach Aktualisierung der Browserseite, oder Neuaufruf keine Verbindung zum Radio (Seite-Ladefehler). Erst nach Reset Radio bekomme ich wieder eine Verbindung
Gruß Sören

schreibfaul1, 2024-11-06:
Ist das jetzt immer noch so? Ich habe noch etwas verändert. Das Problem ist, dass beim Laden der Seite etwa 40 xhr Anfragen gleichzeitig abgeschickt werden, aber nur 36 korrekt beantwortet wurden. Vier blieben ohne Rückmeldung. Das waren jedesmal vier andere. Manchmal wurde dann die Grafik von einem Button nicht geladen.  Ich glaube, das habe ich jetzt gelöst.

skassel2024, 2024-11-07:
Hallo Wolle,
nach Update auf die aktuellste Version 3_5v funktioniert es. Auch ist die Reaktionszeit der Weboberfläche um einiges schneller. Beim Ändern der Senderliste wird diese auch auf der SD sofort aktualisiert, was vorher teilweise mit Fehlermeldung erfolgte.
Eine Fehlermeldung habe ich aber noch beim Aufruf des Players sowohl aus Weboberfläche als auch über TFT:
[image]

Danke nochmals für Deine großartige Arbeit - Respekt!
Gruß Sören

schreibfaul1, 2024-11-08:
Hallo Sören,
danke für deine Rückmeldung.
Den Fehler "getCouloredStringByIndex() idx out of range" habe ich inzwischen eingegrenzt. Hat keine großen Auswirkungen auf die Funktion und wird demnächst beseitigt.
Du hast recht mit deiner Vermutung. Der letzte gespielte Titel soll in der Liste anhand der Position farbig markiert werden.
Das klappt nicht, wenn der Ordner gewechselt wird.

vG
Wolle

closed 2024-12-24 by github-actions[bot]