schreibfaul1/ESP32-MiniWebRadio #552: sd card issue

Vital1984, 2024-11-06:
hi guys

ich have this problem . and can not figure out whats the problem. try different sd cards different sd card adapter everytime same result.

maybe some one can help ?

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.5
ESP-IDF Version: 5.1.4
MiniWebRadio     Version 3.5t - Nov 04/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8374816 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
E (211) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (211) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  1158][E][SD_MMC.cpp:190] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed

kpbicka, 2024-11-06:
Если есть разные шилды картридеров, попробуй их - на всех стоят разные резисторы подтяжки
Для разных говённых микроСД карт может помочь снижение скорости SPI
Возможно длинные провода до ридера
Посмотри картинки от wolle - там нарисованы SMD-резисторы которые или надо удалить или наоборот установить

schreibfaul1, 2024-11-06:
SD_MMC does not require any external resistors. You can connect the SD card directly to the pins of the ESP32.
Some displays with a built-in SD card adapter have resistors that are connected in series. These should be bridged.

Vital1984, 2024-11-06:
hi danke ja das habe ich auch schon versucht die sd karte direkt an den esp32 ohne adapter und ähnliches leider ohne erfolg. jetzt habe ich irgend einen den ich noch liegen hatte ausprobiert damit geht es nun. habe aber nun andere probleme. weis nicht ob ich einen neuen post dazu aufmachen soll oder lieber hier schreiben .

und zwar bekomme ich kein wifi verbindung ob ich meine zugangsdaten in die common.h schreibe oder auf die sdkarte es verbindet sich nicht zu meinem wlan.

und um mich mit dem Accespoint zu verbinden geht auch nicht .es wird einfach nicht gefunden

schreibfaul1, 2024-11-06:
Dann gehe ich davon aus, dass die SD Karte nicht mehr das Problem ist.

Der ESP32 kommt manchmal mit den Einstellungen des Routers nich klar. 
Ich habe hier eine Fritzbox und wenn WPA2+WPA3 eingestellt ist funktioniert es nicht. WPA2(CCMP) hingegen funktioniert. Vielleicht ist es bei deinem Router ähnlich?

closed 2024-12-23 by github-actions[bot]