schreibfaul1/ESP32-MiniWebRadio #556: Zeitanzeige nicht korrekt

skassel2024, 2024-11-10:
Hallo Wolle,
bei der Zeitanzeige springt die Stunde alle paar Sekunden eine Stunde vor oder zurück. Minute und Sekunde sind ok. Das tritt bei der Uhrzeit im Header in allen Modi auf. Auch bei der Uhr sowohl 7-Segment/Flip und im sleep Modus.
Mir ist das bisher nicht aufgefallen, daher kann ich Dir nicht sagen, ab welcher Version das passiert. Ich glaube aber, dass es in der V 3.5i noch nicht so war. Full Clean und Neucompilierung brachten keine Änderung.
Gruß Sören
[image]
[image]
[image]

valera39, 2024-11-10:
[skassel2024](https://github.com/skassel2024), Confirmed, this problem is sometimes present

schreibfaul1, 2024-11-11:
Hallo,
ich habe nir die dafür zuständige Variable _time_s auf die Überwachung gelegt. Und tatsächlich, manchmal ändert sich die Stunde mehrfach kurz hintereinander, dann wieder lange Zeit gar nicht.
Ich vermute, dass sich die dahinterliegende globale Struktur dafür verantwortlich ist. Ich habe das geändert. Bisher sieht es gut aus.

skassel2024, 2024-11-12:
Hallo Wolle,
danke, sieht wirklich gut aus. Nach Deiner Änderung gab es in den letzten beiden Stunden keinen sporadischen Wechsel der Stunden mehr. Hatte auch die Variable im Monitor angezeigt, zusätlich RSSI, hatte erst die Vermutung, dass mein WLAN das Problem verursacht. Hat sich aber nicht bestätigt. In den Abendstunden habe ich öfter das Problem mit "slow stream", obwohl RSSI stabil zwischen -50 und -58 db ist.
Gruß Sören

schreibfaul1, 2024-11-12:
sehr schön, dann ist wieder ein Fehler beseitigt. Dieser Fehler ist mir vorher nie aufgefallen.
Bei den "slow stream" Stationen habe ich festgestellt dass die meisten einen ping > 120ms haben. Unter 100ms ist es immer gut. Und meistens die Bitrate auch nicht sehr hoch. Das wird am WiFi liegen. Mit einem Ethernet-Modul funktionierts immer. Ich habe leider nicht viel Ahnung von WiFi. Wenn ich etwas auf Verdacht verstelle wird es blos schlimmer.

closed 2024-12-28 by github-actions[bot]