schreibfaul1/ESP32-MiniWebRadio #558: Player Fehler und Neustart

skassel2024, 2024-11-12:
Hallo Wolle,
der Player bricht ab und löst einen Neustart aus.
Liegt das an der Verzeichnistiefe ? Wenn ich den Titel auf der SD in ../audiofiles verschiebe wird der Titel abgespielt.

Player spielt Titel problemlos
17:57:14 webSrv: ...  Play "/audiofiles/On the cover of the Rolling Stones.mp3"

Player Fehler und Neustart
17:58:18 webSrv: ...  Play "/audiofiles/myPlaylist/Dirk/Music/Dr. Hook/Greatest Hits (And More)/On the cover of the Rolling Stones.mp3"

Gruß Sören

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x403852d4  PS      : 0x00060833  A0      : 0x80384f39  A1      : 0x3fcd29c0  
A2      : 0x3c2535f4  A3      : 0x3c25c240  A4      : 0x3c25c238  A5      : 0x00000000
A6      : 0xfffffffc  A7      : 0xb33fffff  A8      : 0x3c25c264  A9      : 0xa8763bd4
A10     : 0x3c25365c  A11     : 0x6c50796c  A12     : 0x6c507998  A13     : 0x3fcbeb40  
A14     : 0x3fcaf624  A15     : 0x00060021  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xa8763bd4  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x403852d1:0x3fcd29c0 0x40384f36:0x3fcd29e0 0x4037923b:0x3fcd2a00 0x4038cced:0x3fcd2a20 0x42016eee:0x3fcd2a40 0x4201710f:0x3fcd2aa0 0x4201ae42:0x3fcd2af0 0x42071676:0x3fcd2b70 0x42072726:0x3fcd3420 0x4201e6a2:0x3fcd3480 0x4208647d:0x3fcd3d20
  #0  0x403852d1 in block_absorb at C:\Users\skass\.platformio\packages\framework-espidf\components\heap\tlsf/tlsf.c:454
      (inlined by) block_merge_next at C:\Users\skass\.platformio\packages\framework-espidf\components\heap\tlsf/tlsf.c:487
      (inlined by) tlsf_free at C:\Users\skass\.platformio\packages\framework-espidf\components\heap\tlsf/tlsf.c:1123
  #1  0x40384f36 in multi_heap_free_impl at C:\Users\skass\.platformio\packages\framework-espidf\components\heap/multi_heap.c:231
      (inlined by) multi_heap_free_impl at C:\Users\skass\.platformio\packages\framework-espidf\components\heap/multi_heap.c:220
  #2  0x4037923b in heap_caps_free at C:\Users\skass\.platformio\packages\framework-espidf\components\heap/heap_caps_base.c:64
  #3  0x4038cced in free at C:\Users\skass\.platformio\packages\framework-espidf\components\newlib/heap.c:39
  #4  0x42016eee in connecttoFS(char const*, char const*, unsigned long) at src/main.cpp:1329
      (inlined by) connecttoFS(char const*, char const*, unsigned long) at src/main.cpp:1316
  #5  0x4201710f in SD_playFile(char const*, unsigned long, bool) at src/main.cpp:1836
  #6  0x4201ae42 in SD_playFile(char const*, unsigned long, bool) at src/main.cpp:1814 (discriminator 7)
      (inlined by) WEBSRV_onCommand(String, String, String) at src/main.cpp:3386 (discriminator 7)
  #7  0x42071676 in WebSrv::handlehttp() at lib/websrv/websrv.cpp:615 (discriminator 1)
  #8  0x42072726 in WebSrv::loop() at lib/websrv/websrv.cpp:802
  #9  0x4201e6a2 in loop() at src/main.cpp:2395
  #10 0x4208647d in loopTask(void*) at C:/Users/skass/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74

schreibfaul1, 2024-11-12:
Hallo Sören,
die Speicherverletzung passiert genau hier:
[image]
Hab noch keine Ahnung was die Ursache ist. 
`webSrv: ... Play "/audiofiles/myPlaylist/Dirk/Music/Dr. Hook/Greatest Hits (And More)/On the cover of the Rolling Stones.mp3"` Vielleicht die Länge des Dateinamens oder der Punkt in Dr. Hook? Ich sachaue mir das an.

schreibfaul1, 2024-11-12:
Hole bitte die KCX Lib neu.
Das mit dem setMode dient dazu, dass beim Neustart der BT-Emitter weiß, ob er im TX oder RX Modus starten soll. So wird immer der letzte Zustand wiederhergestellt.

skassel2024, 2024-11-13:
Perfekt! nach "Full Clean"

closed 2024-12-29 by github-actions[bot]