schreibfaul1/ESP32-MiniWebRadio #559: Auto connect to wifi

clear-sky, 2024-11-12:
Hi Dear Wolle!
Where I live, the Internet is unstable - in the previous versions, when the Internet was disconnected and reconnected, the radio connected to the Internet automatically and continued to work, but in the current version, when the Internet is disconnected and reconnected, an orange error message appears on the screen in orange color
i enabled CONFIG_AUTOCONNECT_WIFI=y in sdkconfig but did not help me
can you suggest any way?

schreibfaul1, 2024-11-12:
Hi clear-sky,
I only recently added this behaviour. This is to protect against an "endless loop" if the URL is not available. If the internet goes down, it has the same effect. I could install a timer that tests every minute, for example, whether the Internet is available again.

clear-sky, 2024-11-13:
Hi! Dear Wolle
Thanks for the answer!
internet quality in most countries is NOT as well as europe! will be thankful to add this timer!

schreibfaul1, 2024-11-14:
Hello clear-sky, if the connection is interrupted due to poor internet, a new connection attempt is now started after 30 seconds. The prerequisite is that the local WiFi network is still active.

clear-sky, 2024-11-15:
Hi Dear Wolle!
problem solved !
_Thank you for your attention and your efforts_

closed 2024-11-15 by clear-sky