schreibfaul1/ESP32-MiniWebRadio #560: No questions asked, just gratitude.

valera39, 2024-11-14:
Dear Wolle!
Thank you for your project . Built another Mini Web Radio, this time with a 2+1 amplifier. Thank you for your support!
[image]
[image]
[image]

schreibfaul1, 2024-11-14:
Very neat and accurate work! I'm not that good at craftsmanship. If you agree, I'll add your photos to the gallery.

valera39, 2024-11-14:
Of course I agree.

valera39, 2024-11-19:
[pisascha](https://github.com/pisascha), The entire case is homemade, the wood parts are CNC machined, the front panel is PVC plastic, handmade

pisascha, 2024-11-19:
Very nice 👍🏻 good work

closed 2025-01-04 by github-actions[bot]