schreibfaul1/ESP32-MiniWebRadio #561: NOT ISSUE only some suggestions

clear-sky, 2024-11-21:
Hi Dear Wolle!
I have some suggestions to improve the radio menu with ir remote
1-In all the receivers that I have seen so far, the up and down keys are used to change the channel and the left and right keys are used to change the volume.
2-There is no need for many changes in the menu design, I have two ideas:
a:Assuming we are in radio mode, by pressing the OK button, the channel list will open In this case, it is better to add a part called control panel to the radio where the settings can be changed
b:In the second case, pressing the OK button opens the touch menu as if the user has touched the screen In the next step, the user moves between the icons with the left and right buttons of the remote control. This is possible by changing the color of the icons, and by pressing the OK button, the icon is executed on the screen.
3-In the current version, it is possible to change the brightness of the screen only in touch mode, and it is better to add this option to both the web user interface and the remote control mode.
4-The last suggestion is very difficult and requires a lot of work
Information such as station number- IP and Wi-Fi power levels are useful in radio mode, but they are not very effective in other modes. In other modes, they can be removed and the screen space can be used more optimally.

_This radio is a miracle, it's hard to believe. We are all very grateful for your efforts and we wish you the best from afar._

clear-sky, 2024-11-21:
using another font (specially mono fonts like xolonium) in header and footer section will make the radio  more attractive
In clock mode, it is better to use this font instead of seven segment numbers
https://www.dafont.com/xolonium.font

clear-sky, 2024-11-21:
The seven-segment font is related to the age of dinosaurs

schreibfaul1, 2024-11-21:
Hello clear-sky,
point 1 is easy to realise,
point 2 "ok" jumps to the channel list is also easy. To use the buttons with the remote control I need another colour and cannot be changed so quickly.
Point 3: later all functions will be controllable with the remote control, including the display brightness. 
Point 4 must also be postponed until later. Mono fonts are certainly quite good in some situations, but for the time being it remains with the proportional fonts.

The seven segment display comes from a graphic that is stored on the SD card.  These are in the folder "digits" and "digits_small". If you have any good suggestions please present them here.

closed 2025-01-06 by github-actions[bot]

clear-sky, 2025-02-28:
Hi Dear Wolle!
Now you are working on LVGL
I think this is the time for what i mean: is lvgl strong enough to process big fonts instead of using an image for each number???
if the answer is yes please tell me to send you some nice fonts for clock fonts

schreibfaul1, 2025-03-01:
Hi @clear-sky,
I didn't use LVGL for two reasons,
1) the studio to create the layout is not free for larger designs,
2) I didn't manage to create buttons with a graphic, only text works and all icons would be omitted.

That's why it continues as before with self-created graphic objects. With the large displays, an even larger font would exceed the 4MB Flash. Not everyone has 8MB or 16MB.
But other images would be possible.

clear-sky, 2025-03-02:
Hi Dear Wolle!
thanks for the answer
P4 is comming! 
as far as i know P4 will not come with 4MB flash-minimum amounts for sram and flash will be 8MB!