schreibfaul1/ESP32-MiniWebRadio #563: IR Signal der FB wird nicht immer erkannt

skassel2024, 2024-11-21:
Hallo Wolle,
das IR Signal der FB wird nicht immer erkannt. Im Abstand von wenigen cm funktioniert es besser, obwohl die LED am Empfänger ein Signal, auch bei 4m Abstand anzeigt (Bild). Habe es im Versuchsaufbau mit IR_PIN, INPUT in main.cpp und mit und ohne externem Pullup-Widerstand (47k) probiert. Aber keine Besserung. Aus unerklärlichen Gründen funktioniert es manchmal. Aber meist dann nur einmal, ein weiterer Tastendruck wird dann nicht erkannt. (Timing?, Impulslänge in Library?) 
Gruß Sören

clear-sky, 2024-11-21:
Hi Dear [skassel2024](https://github.com/skassel2024)
#400

clear-sky, 2024-11-21:
Hi Dear [valera39](https://github.com/valera39)
i have used arduino ir reciever (vs838) in other projects based on esp32 with wifi enabled which works easily from a distance of 10 meters!
but in MWR does not work well- wolle has solved this problem with shocking mcu with signals that have 5v edges- technically this way is not correct and problem is in software

skassel2024, 2024-11-22:
Thanks clear-sky, valera39, Wolle for the detailed explanation. I must have overlooked it. Now it works really well with the TSOP4838. No problem up to 6m, even in the immediate vicinity of the WIFI antenna also with 3,3V or 5V VSS

closed 2025-01-06 by github-actions[bot]