schreibfaul1/ESP32-MiniWebRadio #564: SD funktioniert nur mit 10Mhz

skassel2024, 2024-12-04:
Hallo Wolle,
ich weiss, das Problem wurde hier schon des öfteren diskutiert. Leider komme ich aber nicht mehr weiter. Bei meinem letzten Radio (eyeTEK Röhre) funktioniert alles soweit bestens.
Der letzte Versuchsaufbau mit minimalster Verkabelung und den selben Komponenten (ESP32-S3, TFT mit SD) will einfach nicht funktionieren. Wenn ich die SDMMC Speed auf 10Mhz reduziere, wird die SD-Karte erkannt und auch gelesen und es funktioniert. Bei höherer Taktrate wird die Verzeichnisstruktur beschädigt und es erscheint Buchstabensalat, wie hier schon beschrieben.

Was bedeutet SD_MMC  / is not a directory, woher kommt die "STA" Meldung?

Ausgeführt: Erase Flash, .platfformio packages neu geladen, Full Clean. das hat alles nicht weitergeholfen.

Was bedeuen die gelben Zahlen im Player beim Filename? Du hast das mal irgendwann eingefügt, aber ich kann es nicht mehr finden. Das hat aber nichts mit dem SD Problem zu tun, ich möchte es nur verstehen.

Gruß Sören
[image]
[image]

schreibfaul1, 2024-12-04:
Hallo Sören,
ich vermute, die Leitungskapazität wird eine Rolle spielen. Es ist besser, statt dem Bandkabel drei einzelne Kabel zu verwenden.

Die Meldung "STA alredy disconnected..." kommt zusammen mit WiFiMulti.run() aus dem Arduino Framework. Ich habe keine Idee warum das kommt noch wie man diese Meldung unterdrücken kann.

skassel2024, 2024-12-04:
Hallo Wolle,
danke für die schnelle Rückmeldung. auch mit einzelnen Kabeln und kürzeren geht es nur mit 10Mhz. Mehr geht nicht. Bei meinem anderen Radio ist auch Flachkabel 20cm und es geht mit 80Mhz. Allerdings in der Version 3_5w. Aber die Meldung [vfs_api.cpp:28] open():  does not start with / kann ich mir nicht erklären, obwohl die SD gelesen wird.
Gruß Sören
[image]
[image]

muhackl, 2025-01-08:
Hallo zusammen. Ich schließe mich dem Thema an:
- ESP32 S3
- MiniWebRadio V. 3.0.1
- Ich nutze den SD-Adapter - siehe Bild im Anhang.
- Ich erhalte die Fehlermeldung (siehe Bild im Anhang): 
00:00:00 setup: ....  Init SD card
E (2559) sdmmc_common: sdmmc_init_cid: all_send_cid returned 0x107
E (2559) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  2568][E][SD_MMC.cpp:295] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed

- Die Pins des SD_Adapters sind folgendermaßen verbunden: 3,3V->3,3V /// GND->GND /// MOSI->GPIO14 /// MISO->GPIO11 /// CLK->GPIO13
- Ich habe es mit 80 MHz, 40 MHz, 20 MHz, 10 MHz und 5 MHz versucht. Kein Erfolg.

Was mache ich falsch?
Vielen Dank im Voraus für die Hilfe.

[image]
[image]
[image]

Arne65, 2025-01-08:
Hallo muhacki,

auf deinem Adapter scheinen 10Kohm Pul up Widersände zu sein MOSI,MISO und CLK die sollten da raus .

Wenn ich da auf dem Bild richtig deute steht da 103 drauf also 5% 10 kohm 

Denke ich

vG Arne

muhackl, 2025-01-08:
Hi Arne,
ich reiße die mit Vergnügen heraus. Aber in der Anleitung steht, dass das Entfernen bei einem ESP32 S3 nicht notwendig wäre. 
Sicherheitshalber nachgefragt: Die Widerstände entfernen? Oder die Widerstände entfernen und die Kontakte brücken?
VG
Muhackl

Arne65, 2025-01-08:
Da ich jetzt  nicht sehe die Widerstände verbaut sind ,

Sollten die in Reihe sein auf jeden fall raus und brücken .

skassel2024, 2025-01-08:
Hallo muhacki,
bei "Pullup" nur entfernen, ansonsten "in Reihe" brücken. Siehe auch  DLNA-Client Compiling Error #565. Da hatte ich das Problem, dass es erst funktioniert hat, nachdem ich die Anschlüsse verlötet hatte und nicht auf Steckbrett aufgebaut.
Gruß Sören

muhackl, 2025-01-08:
Hi zusammen. Ich habe nochmal versucht, es näher zu fotografieren. Klingt, als ob ich sie nur herausnehmen und nicht brücken müsste, oder?
Alle oder nur MISO, MOSI, CLK?

[image]
[image]

Arne65, 2025-01-08:
mal Messen 

in Reihe =brücken 
Pullup  (also zu GND) raus

skassel2024, 2025-01-08:
bei 10k sind es sicherlich "Pullup" (@Arne65, Pullup zu 3,3V, Pulldown zu GND) also alle 4 nur auslöten. Die beiden Kondensatoren C1 u. C2 dienen der "Stabilierung/Störunterdrückung" der 3,3V Versorgung.
Die Widerstände auf den Displays zu SD-Slot haben meistens 1k. Diese sollten gebrückt werden, wenn der Slot auch verwendet  wird.
Gruß Sören

Arne65, 2025-01-08:
Hallo Sören ,

stimmt .Asche auf mein Haupt . :)

valera39, 2025-01-08:
Hello [muhackl](https://github.com/muhackl)!
I use the exact same adapter as yours.
I have only the MISO resistor unsoldered and everything works without problems, although I had the same problem at first. Also check that the connections are secure

muhackl, 2025-01-08:
Dann taste ich mich mal vor und löte die Widerstände nach und nach aus und beginne mit MISO. Danke für die Tipps

muhackl, 2025-01-08:
Ok es ist recht sicher etwas ganz anderes. Ich habe jetzt erst den Widerstand von MISO, dann nach und nach alle anderen ausgelötet und zwischendrin immer wieder geprüft, ob nun der Fehler nicht mehr auftaucht. Er taucht auch auf, nachdem alle Widerstände raus sind. 
E (504) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x109
E (504) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
[   512][E][SD_MMC.cpp:295] begin(): Failed to initialize the card (0x109). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed

D.h. es ist was anderes. 

Vielleicht nochmal die Frage: Habe ich es denn überhaupt richtig angeschlossen?

3,3V->3,3V 
GND->GND 
MOSI->GPIO14 
MISO->GPIO11 
CLK->GPIO13

Ich möchte es gerne ohne Display betreiben - von daher die Anschlussfrage: Welcher Micro-SD-Karten-Adapter funktioniert direkt (ohne Auslöten von Widerständen usw.)?

Arne65, 2025-01-08:
Das sieht aber schonmal anders aus 
du hast jetzt 109 :)
welche Frequenz hast du jetzt für die Karte ? geh mal zum Test aus 10 MHz

Arne65, 2025-01-08:
https://www.az-delivery.de/products/micro-sd-card-adapter-shield

der sollte funktionieren

muhackl, 2025-01-08:
Hin und wieder kommt auch diese Fehlerkonstellation, aber ich erkenne noch keinen Zusammenhang. Ich teste jetzt nochmal alle Taktfrequenzen durch.
E (5203) sdmmc_periph: sdmmc_host_clock_update_command(148): sdmmc_host_start_command returned 0x107
E (6204) sdmmc_periph: sdmmc_host_clock_update_command(148): sdmmc_host_start_command returned 0x107
E (6204) sdmmc_req: handle_idle_state_events unhandled: 00001000 00000000
E (6210) sdmmc_req: handle_idle_state_events unhandled: 00001000 00000000
E (10218) sdmmc_periph: sdmmc_host_clock_update_command(148): sdmmc_host_start_command returned 0x107
E (10218) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (10224) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[ 10237][E][SD_MMC.cpp:295] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed

Arne65, 2025-01-08:
Da du unterschiedliche Fehler erhältst 107 ,109 eventuell auch 108

es genügen sehr kleine Kapazitäten um das aus dem Tritt zu bringen  

achte auf kurze Leitungen und  ordentlichen Kontaktgabe reinige eventuell deine Lötstellen 
wichtig ist ebenfalls die Spannungsversorgung

skassel2024, 2025-01-08:
oder sowas:

[image]

Anschluss lt. Wolles Beschreibung:
[image]

möglichst kurze Verbindung und kein Steckbrett. Damit habe ich lange den selben Fehler gesucht. Am besten verlöten.
Gruß Sören

muhackl, 2025-01-08:
Ich denke, es ist etwas ganz anderes.
Werder mit 5, 10, 20, 30, 40 oder 80 MHz funktioniert es.
Stimmen denn die Anschlüsse? Habe ich die Skizzen richtig interpretiert für den ESP32 S3?
3,3V->3,3V
GND->GND
MOSI->GPIO14
MISO->GPIO11
CLK->GPIO13

muhackl, 2025-01-08:
Die Verbindungen sind direkt Verlötet. Die Kabel sind ca. 5 cm lang. Das Board steckt logischerweise am USB-Anschluss des PCs, da ich ja sonst die Console nicht sehen kann. Aber ich teste es nochmal mit einem 5V-Netzteil.

Arne65, 2025-01-08:
ich meinte weniger die Stromversorgung des Board sondern eher die 3,3V an deinem Adapter

ja das passt mit Deinen Anschlüssen

eventuell liegt es einfach wirklich am Adapter oder der Karte   (Kontakte)

valera39, 2025-01-08:
[muhackl](https://github.com/muhackl),, Yes you have the correct connection, perhaps the problem in the adapter contacts, although I repeat I have three such adapters with removed resistors on MISO with a length of wires 15 cm work normally at maximum frequency. Try making a homemade adapter from a micro SD adapter.
Also try another memory card

muhackl, 2025-01-10:
@Arne65 die hier https://www.az-delivery.de/products/micro-sd-card-adapter-shield haben ja gar kein CMD und kein CLK. Oder welche der Pins entsprechen CMD und CLK?

Arne65, 2025-01-10:
Belegung:
D8 - SS
D7 - MOSI
D6 - MISO
D5 - CLK

muhackl, 2025-01-10:
Ja. Ich hätte vorher nach unten scrollen sollen ;-)
Danke.

muhackl, 2025-01-10:
Und schon ist der Spaß wieder vorbei. Nachdem der ESP ca. 2 mal gestartet ist, so dass er wie oben beschrieben zwar die SD-Karte erkannt, aber die Weboberfläche nicht wirklich erreichbar war, ist nun wieder die SD-Karte mit dem altbekannten fehler nicht mehr erkennbar:
E (212) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x109
E (213) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
[   222][E][SD_MMC.cpp:295] begin(): Failed to initialize the card (0x109). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed

Geändert habe ich NUR den WIFI_TX_POWER-Wert, um eine bessere WLAN-Verbindung zu erreichen. Mit Werten, die vormals wenigstens noch die Grundfunktionalität erlaubten, geht es nun auch nicht mehr. 
Ich gebe auf.

Arne65, 2025-01-10:
Deine SD Karte wird jetzt leer sein oder das Dateisystem ist beschädigt 
Formatieren und neu beschreiben.

Deine Probleme liegen mit Sicherheit an deiner Verkabelung 

Ich kann dir nur sagen wenn alles stimmt 
-Kontakte und Spannung funktioniert das wie geschmiert 

Die Webseite wirst du nicht richtig erreichen weil dein WLAN an dem Ort an dem du das Radio betreibst etwa dünn zu sein scheint oder von Geräten in der Nähe beeinflusst wird
weiterhin solltest du darauf achten nur ein Fenster auf hast mit der Radio IP

Arne65, 2025-01-10:
Eventuell bricht deine 3,3 V  für die SD Karte Stromversorgung etwas ein

muhackl, 2025-01-10:
Ich würde sagen, das ist dann absolut zu empfindlich. 
Es ist mit Standard-Kabeln, die ich bereits in unzähligen ESP- und SBC-Projekten verwendet habe,  ganz normal verlötet. 
Stromversorgung ist ein 5V 2A Netzteil.
Der ESP liegt 20 cm vom WLAN-Repeater entfernt. Ja, es sind 5 andere Geräte in der Nähe, aber wenn es das nicht abkann - schwierig.
Die Daten sind noch auf der SD-Karte - ich hab sie formatiert und nochmal beschrieben, danach ging es wieder gar nicht mehr, nachdem es vorher funktionierte. 
Da ist etwas anderes im Argen. Wahrscheinlich einfach ein Grund-Defekt am ESP. Ich gebe dem Webradio einen letzten Versuch mit einem "normalen" ESP (Nicht-S3) und einer popeligen Verkabelung mit Jumper-Kabel und Metall-Pins.

muhackl, 2025-01-12:
Mein "normaler" ESP32 ist ein WROOM und hat keinen psram. Wun - der - bar. In der plattformio.ini "-D BOARD_HAS_PSRAM=0" zu setzen bringt leider nichts. Es soll einfach nicht sein.

[image]

muhackl, 2025-01-12:
Ein letzter Versuch mit dem ESP32 S3 und der Verbindung des SD-Karten-Adapters per billigster Jumper-Kabel und den angelöteten Pins hat den Erfolg gebracht. FAST. Es geht nun hier weiter, denn eine Fehlermeldung blockiert den ESP nun: https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/566

Kicadder, 2025-01-14:
Your problems look like power supply instabilities. You could try to stabilize the power supply lines, like Arne65 did in his (excellent) PCBs : 0.1µF in parallel with 10µF at the  nearest place of every PCB's power line (5V or 3.3V) : ESP32, display, SDCard, etc.
Maybe trying another power supply could solve also the problem...
NB : capacitors values are not critical at all

closed 2026-04-07 by github-actions[bot]