schreibfaul1/ESP32-MiniWebRadio #565: DLNA-Client Compiling Error

skassel2024, 2024-12-05:
Hallo Wolle,
die aktuelle Version lässt sich nicht compilieren. Das ist mir aufgefallen, als ich wegen des SD Problems das ganze neu installieren wollte.
Gruß Sören
[image]

hevet, 2024-12-05:
Use the DLNA library and install new versions.

schreibfaul1, 2024-12-05:
Hallo Sören,
hast du das Problem mit der SD Karte lösen können? Eine gegenseitige Beeinflussung zwischen Daten- und Taktleitung wäre due wahrscheinlichste Ursache gewesen, dafür reichen schon wenige pF bei 80MHz. Bleibt nur noch irgendwo eine schlechte Kontaktgabe, Übergangswiederstände oder zu niedrige Spannung an der SD Karte.

Zu dem anderen Fehler, strncpy ist ein Standard C Kommando. Ich glaube die Ursache liegt irgendwo anders. Manchmal ist die Meldung vom Kompiler nicht eindeutig. In der DLNA Bibliothek hat es eine Änderung gegeben, aber an anderer Stelle. Wenn die Kette "HTTP Request -> Response-> Payload" unterbrochen ist, wird die Anfrage wiederholt und ist somit zuverlässiger. Du kannst in .pio/libdeps/esp32s3/ den Ordner DLNA-Client löschen, dann wird automatisch die neue Version geladen.
vG Wolle

schreibfaul1, 2024-12-05:
Die KI meint, ich sollte statt strncpy lieber strlcpy nehmen, dann würde die Funktion so aussehen:
````c++
char* x_ps_strndup(const char* str, uint16_t len) {
    if (!str) {  log_e("given str is NULL");  return NULL; }
    size_t str_len = strlen(str);
    if (len > str_len) len = str_len;

    char* ps_str = NULL;
    if (m_PSRAMfound) { ps_str = (char*)ps_malloc(len + 1); }
    else              { ps_str = (char*)malloc(len + 1); }

    if (!ps_str) { log_e("oom");   return NULL; }
    strlcpy(ps_str, str, len + 1); // len+1 garantiert Nullterminierung
    return ps_str;
}
````

skassel2024, 2024-12-06:
Hallo Wolle,
SD Problem noch nicht gelöst. Leitungen gekürzt, Kontakte ok, Spannung 3,29V aber bei 10Mhz ist Schluss. Werde da aber weiter dran bleiben.
DLNA wird zwar nach dem Löschen neu geladen, aber der selbe Fehler (strncpy). 
Erst nach manuellem Einfügen vom Script der "KI" funktioniert es. 

[image]

Nach Full Clean und Neucompilierung ist der Fehler wieder da. 
 In [ESP32-DLNA-Client](https://github.com/schreibfaul1/ESP32-DLNA-Client) ist das nicht geändert?
[image]

skassel2024, 2024-12-06:
Hallo Wolle,
soeben erstaunliche Erkenntnis. Nach Neucompilierung und Neustart ohne Änderung an der Hardware gehen plötzlich 80Mhz - bin "konfused" ???
Was bleibt, ist noch die Fehlermeldung: 
[  1448][E][vfs_api.cpp:28] open():  does not start with / 
SD_MMC/ is not a directory
Gruß Sören
[image]

schreibfaul1, 2024-12-06:
Das ist eigenartig. Solche Fehler kommen bestimmt wieder. Vielleicht ist es ein Kontaktproblem innerhalb des Kartenadapters?
strncpy habe ich durch strlcpy ersetzt. Warum der Kompiler bei dir an der Stelle meckert ist mir unklar, zumal es strncpy im Projekt noch an weiteren 16 Stellen gibt.
[image]
gibt es an zwei Stellen, dievom setup überhaupt nicht aufgerufen werden. Schließe mal bitte das Terminal und führe ein "Erase Flash" aus.
Oder das Framework ist kaputt. Du könntest den neuen Branch "new-IR-menue" testen, das benutzt eine neuere Arduino und IDF Version.

skassel2024, 2024-12-06:
Hallo Wolle,
Problem SD gelöst. zu meiner Schande muss ich gestehen, es lag am Steckboard. SD und ESP verlötet funktioniert nun zuverlässig mit 80Mhz, unabhängig der Arduino und IDF Version. 
[image]

Die andere Meldung 
[   430][E][vfs_api.cpp:28] open():  does not start with /
SD_MMC/ is not a directory
besteht weiterhin, auch under Arduino 3.1.0 und ESP-IDF 5.3.1
Was mir aber aufgefallen ist, **nur** wenn ich "Content_on_SD_Card" direkt auf die SD entpacke, kommt die Meldung nicht mehr. Sobald ich sowohl am PC über SD-Adapter oder per ftp Dateien übertrage (z.B. eigene stations.json, ir_buttons.json), kommt die Meldung wieder. Auch wenn ich die entpackten dateien aus "Content_on_SD_Card" nur auf die SD kopiere.
Gruß Sören 
[image]

skassel2024, 2024-12-08:
Hallo Wolle,
wenn ich in "defaultsettings()" Zeile518-523 ausblende, erscheint die Meldung "SD_MMC/ is not....." nicht mehr. Um das zu verstehen, fehlen mir aber die programmierkenntnisse.
Gruß Sören
[image]

schreibfaul1, 2024-12-08:
Danke, jetzt konnte ich den Fehler reproduzieren. 
lastconnectedfile ist ein Wert in settings.json und soll beim Öffnen der Dateiliste den aktuellen Ort der Datei anzeigen.
In der nächsten Version ist der Fehler beseitigt.

closed 2025-01-23 by github-actions[bot]