schreibfaul1/ESP32-MiniWebRadio #566: DLNA Client

Arne65, 2024-12-08:
Hallo Wolle ,

habe folgende permanente Ausgabe 

[394642][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[394653][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[394706][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[394717][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[394745][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[394756][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[394809][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[394820][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[394839][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[394850][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[394875][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[394886][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[394905][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[394916][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[395007][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[395018][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[395114][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[395125][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding
[395146][E][NetworkClient.cpp:272] connect(): socket error on fd 61, errno: 104, "Connection reset by peer"
[395157][E][DLNAClient.cpp:141] srvGet(): The server 192.168.2.171:2869 is not responding

woher kommt das ? 

Weiterhin ist mir folgendes aufgefallen sofern man einen ESP S3 N8R2 benutzen will muss der Typ vom PSRAM umgestellt werden .Ist das gewollt ?

Viele Grüße Arne

TubaFonse, 2024-12-09:
Hallo zusammen! 
Ergänzung zur Fehlersuche, diese
errno: 104, "Connection reset by peer" 
erhalte ich auch sporadisch, sogar wenn ich nur über das Web-Interface die Lautstärke verändere, d.h. der aktuelle Status auf die SD-Karte geschrieben werden soll: 
[image]
16:05:47 StationName: - 0 N - Evergreens on Radio
16:05:47 icy-url: ..  http://www.0nradio.com
16:05:47 AUDIO_info:  AACDecoder has been initialized, free Heap: 153143 bytes , free stack 3136 DWORDs
16:05:47 lastURL: ..  https://0n-evergreens.radionetz.de/0n-evergreens.aac
16:05:47 AUDIO_info:  stream ready
16:05:47 AUDIO_info:  syncword found at pos 0
16:05:47 AUDIO_info:  Channels: 2
16:05:47 AUDIO_info:  SampleRate: 44100
16:05:47 AUDIO_info:  BitsPerSample: 16
16:05:47 AUDIO_info:  BitRate: 28940
16:05:47 AUDIO_info:  AAC HeaderFormat: ADTS
16:05:47 AUDIO_info:  Spectral band replication: upsampled SBR
16:05:47 StreamTitle: Camouflage - Love Is A Shield
16:05:48 Audiocodec:  AAC
[950387][E][NetworkClient.cpp:433] write(): fail on fd 62, errno: 104, "Connection reset by peer"
16:06:32 webSrv: ...  File not found "/png/Button_Volume_Up_Yellow.png"

d.h. der Fehler erscheint sporadisch, obwohl die Dateien ordnungsgemäß auf der SD-Card existieren (Laufzeitproblem?)

Gruß, Alfons

TubaFonse, 2024-12-09:
[image]

Arne65, 2024-12-09:
Hallo Wolle

es war einer meiner Rechner verursacht .

Danke Gruß Arne

Arne65, 2024-12-09:
Hallo Wolle ,

einer meiner Rechner hat das Problem verursacht.

Danke 
Gruß Arne

muhackl, 2025-01-12:
Ich schließe mich hier an. Nach meiner Odysee hier https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/564 erhalte ich nun diese Meldung im Sekundentakt und der ESP 32 S3 bremst sich damit selber aus.

[ 14776][E][DLNAClient.cpp:141] srvGet(): The server 0.0.0.0:1901 is not responding
[ 17794][E][DLNAClient.cpp:141] srvGet(): The server 0.0.0.0:1901 is not responding
[ 20810][E][DLNAClient.cpp:141] srvGet(): The server 0.0.0.0:1901 is not responding
[ 23831][E][DLNAClient.cpp:141] srvGet(): The server 0.0.0.0:1901 is not responding
[ 26848][E][DLNAClient.cpp:141] srvGet(): The server 0.0.0.0:1901 is not responding
[ 29869][E][DLNAClient.cpp:141] srvGet(): The server 0.0.0.0:1901 is not responding

muhackl, 2025-01-17:
Ich habe heute nochmal einen komplett neuen ESP32 S3 mit dem aktuellen Code aus dem Github Repo bespielt. Keine Veränderung. Er blockiert sich ohne Ende selbst. Sinnvoll wäre, wenn er irgendwann mal aufgeben würde. Aber es läuft absolut endlos die Fehlermeldung durch. Vom Aufbau der Weboberfläche, die man währenddessen versucht, aufzurufen, hätte ich ein Video machen sollen. Da sieht man wunderbar, wie sich der ESP so selbst blockiert: Fehlermeldung, Icon lädt, Fehlermeldung, Icon lädt, Fehlermeldung, Icon lädt usw.
Gibt es schon eine Erkenntniss, wie man das umgehen oder deaktivieren kann?

muhackl, 2025-01-20:
Ergänzung: Ich habe den ESP nun mit einem anderen Netzwerk verbunden, in dem er das einzige Gerät ist. D.h. offenbar gab es in dem anderen Netzwerk ein DLNA-fähiges Gerät, mit dem er sich auf Biegen und Brechen versucht hat, zu verbinden. 
Drei Sachen sind nicht sinnvoll bzw. fraglich:
- warum lautet die Adresse 0.0.0.0:1901 - das klingt so, als ob er sich mit sich selber verbinden möchte. Offenbar ist das im neuen Netzwerk nicht mehr der Fall. Ich versuche, das entsprechende Gerät nun mit Hilfe des Ports ausfindig zu machen.
- warum blockiert er sich damit selbst und bricht nicht nach X Versuchen ab oder gibt eine Statusmeldung über die Weboberfläche aus, die über das weitere Vorgehen entscheiden lässt?
- warum kann man zu DLNA grundsätzlich keine Einstellungen vornehmen - z.B. deaktivieren?

schreibfaul1, 2025-01-20:
Das ist eigenartig.
Beim Start wird ein Rundspruch per udp ins Heimnetz geschickt. Dann wird eine kurze Zeit auf mögliche Antworten gewartet. Wenn ein DLNA-Mediaserver antwortet wird dessen IP Adresse in eine Liste eingetragen und nach dessen Namen gefragt. Das ist das "srvGet()"
Ich würde gerne wissen was bei dir auf 0.0.0.0 Port 1901 antwortet.
Zur Not könnte diese IP als ungültig erklärt werden, was sie sowieso schon ist.

Das Ergebnis des Rundspruchs sieht bei mir so aus:

[image]

das ist die FritzBox und ein DLNA Server, den ich auf meinem PC installiert habe.

muhackl, 2025-01-20:
Ich konnte es mit Wireshark herausfinden. Es sind zwei Sonos-Lautsprecher. Eigentlich sind das doch DLNA-Clients und keine Server. 
Und seltsam bleibt, dass der ESP eben die 0.0.0.0:1901 zeigt und nicht die IP-Adressen der Sonos-Lautsprecher. 
Wie gesagt: Sobald man die Lautsprecher abstöpselt oder man den ESP mit einem "leeren" WLAN verbindet, gibt es die Meldung nicht mehr. Also sind die beiden die Übeltäter und der ESP kann nicht so gut damit umgehen.

muhackl, 2025-01-21:
Ja, das sieht jetzt besser aus und er scheint die DLNA-Störenfriede mehr oder weniger zu ignorieren und auch etwas darüber zu "erzählen". Siehe Screenshot. 
Die Weboberfläche lädt ohne Verzögerung. Auch scheint er einen Stream zu laden und wiederzugeben. Mein PCM5102 gibt jedoch noch keinen Mucks von sich. Aber das ist wahrscheinlich eine andere Baustelle.

Arne65, 2025-01-21:
Hast du auf dem PCM Modul die richtigen Brücken gemacht siehe 

[image]