schreibfaul1/ESP32-MiniWebRadio #567: Upload failed

TubaFonse, 2024-12-09:
Hallo Wolle, 
wenn ich größere Dateien (z.B. neue Logos oder MP3s) über das Web-Interface auf die SD-Card hochladen will, erhalte ich folgende Fehlermeldung: 
[image]
Bei kleineren Dateien. z.B. Playlists kommt beim ersten Versuch die gleiche Fehlermeldung, beim zweiten Versuch, die gleiche Datei hochzuladen, funktioniert es meistens. Kannst Du das reproduzieren? 
Der Fehler erscheint nicht im Serial Monitor: 
[image]

Gruß Alfons

TubaFonse, 2024-12-09:
Ich konnte den Fehler weiter eingrenzen: 
Nur **.jpg-Dateien** können nicht hochgeladen werden! 
Bitte testen, 
Danke & Grüße, Alfons

schreibfaul1, 2024-12-09:
Hi Alfons,
ich konnte durch deine Hilfe den Fehler gut eingrenzen. Das ist in der nächsten Version bereinigt (jetzt zunächst im "new-IR-menue" Branch).
vG
Wolle

TubaFonse, 2024-12-09:
So habe ich inzwischen durch Trick/Workaround den Fehler umgangen: Wenn ich die hochzuladende Datei z.B. Sender_xyz.jpg Datei am PC zuvor umbenenne, z.B. in Sender_xyz_jpg.pdf, dann funktioniert der Upload. Mit "rename" kann ich die Datei übers Web-Interface wieder in Sender_xyz.jpg zurück umbenennen. 
Gruß, Alfons

TubaFonse, 2024-12-09:
Danke Wolle, die .jpg Dateien lassen sich jetzt mit der aktuellen Version aus "new-IR-menue" Branch über das Web-Interface auf die SD-Card hochladen. 
VG, Alfons

closed 2024-12-09 by TubaFonse