schreibfaul1/ESP32-MiniWebRadio #569: Play All issue

TubaFonse, 2024-12-09:
Hallo Wolle, 
Das "Play All"-Problem kann wie folgt rekonstruiert werden. 
Ich habe auf meiner SD-Card unter "audiofiles" eine Playlist Eagles.3mu zum Abspielen von MP3s meiner NAS über DLNA sowie ein Unterverzeichnis Blech_MP3 mit paar Dateien angelegt: 
[image]
Mit Rechte-Maustaste Play starte ich vom Web-Interface meine Playlist Eagles.3mu
Während die Titel noch nicht alle abgespielt wurden, starte ich schon mit Rechte-Maustaste **Play-All** das Unterverzeichnis Blech_MP3 um alle darunter enthaltenen MP3s zu spielen. 
Jetzt tritt der "falsche-Verzeichnis-Fehler" auf, dass von **/audiofiles/audiofiles/**...mp3 versucht wird zu spielen, anstatt im tatsächlichen Verzeichnis /audiofiles/...mp3 - das geht natürlich schief. 
[image]
[image]
[image]

Ich habe keine andere Möglichkeit, eine bereits laufende Playlist abzubrechen (mit der Pause-Taste wird der nächste Titel der Playlist gespielt. 
Wenn die Playlist Eagles.3mu bereits vollständig abgespielt wurde und ich erst dann das Verzeichnis Blech_MP3 mit "Play-All" starte läuft es korrekt ab, Verzeichnis /audiofiles/...mp3 
Kannst Du das bei Dir nachvollziehen? Danke & Gruß, Alfons.

TubaFonse, 2024-12-10:
Hallo Wolle, 
die geschilderten Probleme sind mit der aktuellsten Version leider noch nicht behoben - hab es soeben getestet. 
LG Alfons

TubaFonse, 2024-12-11:
Auch über das Touchpanel gibts den Fehler, dass der Player versucht, aus dem "falschen" Verzeichnis SD_MMC//audiofiles/audiofiles zu lesen: 
Serial Monitor: 
13:03:50 AUDIO_info:  Spectral band replication: upsampled SBR
13:03:50 StreamTitle: Chris Howland - Mädchen Für Alles 
13:03:53 click on ..  backpane
13:03:55 click on ..  backpane
13:03:58 click on ..  btn_RA_player
13:03:59 SD_MMC//audiofiles/audiofiles not exist
[4591069][E][common.h:860] getColouredSStringByIndex(): idx 0 is oor, max = -1
13:04:03 click on ..  btn_PL_fileList
13:04:03 SD_MMC//audiofiles/audiofiles not exist
13:04:03 SD_MMC//audiofiles/audiofiles not exist
13:04:08 SD_MMC//audiofiles/audiofiles not exist
[4600655][E][common.h:860] getColouredSStringByIndex(): idx 0 is oor, max = -1
13:04:15 click on ..  btn_PL_fileList
13:04:15 SD_MMC//audiofiles/audiofiles not exist
13:04:15 SD_MMC//audiofiles/audiofiles not exist
13:04:17 click on ..  lst_PLAYER
13:04:25 click on ..  btn_PL_fileList
13:04:28 click on ..  lst_PLAYER

Gruß, Alfons

schreibfaul1, 2024-12-13:
Hab verschiedene Sachen geändet. Die Playlist wird jetzt zuverlässiger gelöscht.
Und ich hoffe, dass '//' wie in `00:10:14 SD_MMC//audiofiles/audiofiles/myPlaylist not exist` nicht mehr auftreten.

TubaFonse, 2024-12-13:
Hallo Wolle, 
der Fehler ist leider noch da, bitte selber ausprobieren, wie oben beschrieben, Gruß, Alfons: 
13:06:57 Playlist:    http://192.168.178.24:50002/m/NDLNA/33081.mp3
13:06:57 AUDIO_info:  buffers freed, free Heap: 152295 bytes
13:06:57 AUDIO_info:  connect to: "192.168.178.24" on port 50002 path "/m/NDLNA/33081.mp3"
13:06:57 AUDIO_info:  Connection has been established in 16 ms, free Heap: 152179 bytes
13:06:57 AUDIO_info:  MP3Decoder has been initialized, free Heap: 150359 bytes , free stack 3156 DWORDs
13:06:57 AUDIO_info:  ID3 framesSize: 1342
13:06:57 AUDIO_info:  ID3 version: 2.3
13:06:57 AUDIO_info:  ID3 normal frames
13:06:57 id3data: ..  ContentType: (12)
13:06:57 id3data: ..  Title: Blaskapelle Gloria - Flashdance
13:06:57 id3data: ..  Artist: Blaskapelle Gloria
13:06:57 Audiocodec:  MP3
13:06:58 AUDIO_info:  Audio-Length: 2941634
13:06:58 AUDIO_info:  Webfile: stream ready, buffer filled in 207 ms
13:06:58 AUDIO_info:  syncword found at pos 0
13:06:58 AUDIO_info:  MPEG-2.5, Layer I
13:06:58 AUDIO_info:  Channels: 2
13:06:58 AUDIO_info:  SampleRate: 44100
13:06:58 AUDIO_info:  BitsPerSample: 16
13:06:58 AUDIO_info:  BitRate: 128000
13:07:15 AUDIO_FILE:  AudioCurrentTime 0:17s, AudioFileDuration 3:04s  
13:07:16 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
13:07:16 AUDIO_info:  syncword found at pos 93
13:07:16 AUDIO_info:  syncword found at pos 0
13:07:16 AUDIO_info:  MPEG-2.5, Layer I
13:07:16 AUDIO_info:  Channels: 2
13:07:16 AUDIO_info:  SampleRate: 44100
13:07:16 AUDIO_info:  BitsPerSample: 16
13:07:16 AUDIO_info:  BitRate: 128000
13:07:28 AUDIO_FILE:  AudioCurrentTime 0:30s, AudioFileDuration 3:04s  
13:07:29 webSrv: ...  GetFolder "/audiofiles/Blech_FLAC"
13:07:31 AUDIO_FILE:  AudioCurrentTime 0:33s, AudioFileDuration 3:04s  
13:07:32 Playlist:    playlist stopped
13:07:32 webSrv: ...  Play Folder"/audiofiles/Blech_FLAC"
13:07:33 Playlist:    /audiofiles/audiofiles/Blech_FLAC/01_Das Tor der Alpen.flac
13:07:33 AUDIO_FILE:  01_Das Tor der Alpen.flac
13:07:33 AUDIO_info:  buffers freed, free Heap: 150603 bytes
13:07:33 AUDIO_info:  ERROR: File doesn't exist: /audiofiles/audiofiles/Blech_FLAC/01_Das Tor der Alpen.flac
13:07:33 action: ...  current volume is 21
13:07:33 AUDIO_info:  next playlist file
13:07:33 Playlist:    /audiofiles/audiofiles/Blech_FLAC/02_Wenn der Mond erzählen könnt.flac
13:07:33 AUDIO_FILE:  02_Wenn der Mond erzählen könnt.flac
13:07:33 AUDIO_info:  buffers freed, free Heap: 150667 bytes
13:07:33 AUDIO_info:  ERROR: File doesn't exist: /audiofiles/audiofiles/Blech_FLAC/02_Wenn der Mond erzählen könnt.flac
13:07:33 Playlist:    playlist stopped
13:07:33 webSrv: ...  Play Folder"/audiofiles/Blech_FLAC"
13:07:33 Playlist:    /audiofiles/audiofiles/Blech_FLAC/01_Das Tor der Alpen.flac
13:07:33 AUDIO_FILE:  01_Das Tor der Alpen.flac

TubaFonse, 2025-01-13:
Problem gelöst, danke!

TubaFonse, 2025-01-13:
gelöst - siehe #594

closed 2025-01-13 by TubaFonse