schreibfaul1/ESP32-MiniWebRadio #571: Nur ein Danke

skassel2024, 2024-12-12:
Hallo Wolle,
an dieser Stelle möchte ich Dir einmal Danke sagen für dein tolles Projekt und für Deine konstruktive Unterstützung. Ich bin jetzt beinahe 70 Jahre alt und hätte nicht gedacht, dass ich dieses Projekt so einfach nachvollziehen kann. Das MiniWebRadio mit dem eyeTEC Röhrenverstärker hat nun einen festen Platz in der Wohnung bekommen.
Ausserdem habe ich dadurch viel gelernt über die Programmiermöglichkeiten von Arduino und ESP.

Ich wünsche Dir eine schöne Weihnachtszeit und noch viele tolle Ideen.
Gruß Sören

mitja1955, 2024-12-12:
Hello,
I have been working with electronics for over fifty years (from vacuum tubes to microprocessor programming and PCB design), but since I retired, I have been using my time to make various hobby devices. One of these is the miniweb radio, which is a great solution for everyone.
Thanks to the developer for the successful project, and I am currently planning a printed circuit board where it will be possible to choose between both versions of the esp32, three different display sizes, a light sensor, an SD adapter, and both DAC decoders, an IR receiver, and Bluetooth devices.
Thank you Wolle, and continue the project successfully..

valera39, 2024-12-12:
Greetings from Russia!
Your work unites many people around the world regardless of nationality. I fully support the previous users and join in wishing you all the best and continuation of the wonderful project!

schreibfaul1, 2024-12-12:
Dear users,
thank you very much for your kind wishes and contributions. I myself have been retired since this year and am only programming this project as a hobby. I have turned down all requests to work with various companies as I prefer to remain independent. The project is now so big that I have around 100 calls from various users worldwide every day. [image] There are many more with the audioI2S library.
Version 3.6 will be released before Christmas. This will allow almost all functions to be controlled with the IR-RC.
Merry Christmas to the entire community from Berlin
Wolle

samand587, 2024-12-13:
Dear author! Thanks for the project! Merry Christmas!!!

closed 2025-01-28 by github-actions[bot]