schreibfaul1/ESP32-MiniWebRadio #576: BT_EMITTER Pindefinition -1 Fehlermeldung

skassel2024, 2024-12-16:
Hallo Wolle,
wenn ich BT mit -1 definiere
[image]

erscheint folgender Fehler
[image]

Gruß Sören

schreibfaul1, 2024-12-18:
Den Fehler konnte ich nicht reproduzieren, obwohl uint8_t x = -1; immer zu  x = 255; führt.
Ich habe auf "Verdacht" etwas geändert, das kann eigentlich nur ein `pinMode(xxxx, IN_OUT);` sein.

skassel2024, 2024-12-18:
Hallo Wolle, 
der Fehler ist weiterhin da. Es scheint etwas mit der IDF Version zu tun haben. Ich habe unter dem folgenden Link etwas gefunden, was mit Pin 19 und 20 und der IDF zu tun hat. Komme aber da nicht weiter.
Der Fehler tritt nur bei BT_EMITTER_MODE und BT_EMITTER_LINK auf. Aber möglicherweise bin ich komplett auf dem falschen Weg.
[image]
[image]

Gruß Sören
http://esp32.io/viewtopic.php?f=13&t=36111&sid=4932462b301951160d4b14e36a49ff94

skassel2024, 2024-12-26:
Frohe Weihnachten Wolle,
ich glaube dem Fehler auf der Spur zu sein. In KCX_BT_EMITTER.cpp habe ich die Zeile 454 "if(!m_f_KCX_BT_Emitter_isActive) return;" eingefügt.
in Zeile 455 oder 456 bei "digitalWrite" wird der Fehler erzeugt, da hier BT_EMITTER_MODE "-1" ist.
Ich hoffe, damit richtig zu liegen?
[image]
Gruß Sören

schreibfaul1, 2024-12-26:
Gleichfalls schöne Weihnachten,
ich habe bisher noch keine Lösung für das Problem gefunden, weil mit die Ursache nicht bekannt ist, es kann sein, dass du richtig liegst, ich schaue mir das genauer an.
VG Wolle

schreibfaul1, 2024-12-27:
Hallo Sören,
[image]
habe ich hinzugefügt und damit wird die Fehlermeldung nicht mehr erscheinen. Der Pin muss nur gesetzt werden wenn ein KCX_BT_Emitter überhaupt existiert. 
beste Grüße
Wolle

closed 2025-02-10 by github-actions[bot]