schreibfaul1/ESP32-MiniWebRadio #577: Stations-List scrolling up (IR Remote-Control)

TubaFonse, 2024-12-17:
Hallo Wolle, 
mit der IR-Fernbedienung funktioniert das Scroll-**Up** in einer Senderliste (Pfeiltaste nach **oben** = zurückblättern von einzelnen Sendern) nicht über **mehrere Seiten**, sondern nur innerhalb der aktuellen Seite, auf der man sich befindet - blättert man von Sender Nr. 15 schrittweise zurück und ist in der Liste oben am Anschlag, kommt man nicht mehr auf die erste Seite zurück. 
Im Player für Audiofiles funktioniert jedoch das Scroll-Up auch über mehrere Seiten. 

Das Scroll-Down in der Senderliste funktioniert jedoch perfekt, d.h. schrittweises Vorwärtsblättern: Wenn mit der Pfeil-Taste nach unten die unterste Zeile auf einer Seite erreicht ist, wird automatisch zur nächsten Seite gesprungen. Nur zurück gehts halt nicht über mehrere Seiten. 

Viele Grüße, 
Alfons.

schreibfaul1, 2024-12-18:
Hallo Alfons,
das funktioniert bei mir auch nicht, ich werde mir das demnächst ansehen.
vG Wolle

schreibfaul1, 2024-12-18:
Jetzt sieht es besser aus.

TubaFonse, 2024-12-18:
... bei mir funktioniert's auch 👍 
vielen Dank für die schnelle Korrektur! 
LG, Alfons

closed 2024-12-18 by TubaFonse