schreibfaul1/ESP32-MiniWebRadio #579: threading problems Radio Spinner

valera39, 2024-12-24:
Hello Dear Wolle and to all community members
Happy holiday to everyone who celebrates Christmas!
Please check the playback of streams from Spinner radio, for example 
https://live.radiospinner.com/clsscrdsc-96
https://live.radiospinner.com/sssrht-96
https://live.radiospinner.com/elecpop-96
Playback often stops. There are no problems with streams from other stations.
Thank you!

schreibfaul1, 2024-12-25:
Hi valera39,
I have changed something in the sdkconfig to request lost data packets again. It now works quite well with the new IDF 5.3.1. However, there is a problem with live.radiospinner.com from my location, the connection is regularly interrupted (approx. every 10 minutes) and then there is no more ping and only HTTP404 is received.

valera39, 2024-12-25:
Thanks, I had the radio playing for 30 mins. before the 404 error appeared
Is it me or is it just me or has the new version of IDF got wifi connection problems again? Very often an error about no wifi networks is displayed and the radio switches to a hotspot. If you manually enter the name and password, the radio does not see the network again when rebooting. But after several reboots the connection is successful

closed 2025-02-08 by github-actions[bot]