schreibfaul1/ESP32-MiniWebRadio #580: No StreamTitle on Display after "Advertising xx Seconds" displayed

TubaFonse, 2024-12-24:
Lieber Wolle, 
wenn ich auf einen anderen Sender umschalte, kommt ja bei vielen Sendern ein (nerviger) Werbevorspann, bevor das eigentliche Programm gespielt wird. 
Nachdem jedoch die Werbung (Advertising xx Seconds) vorbei ist, wird nicht nicht der StreamTitle angezeigt, sondern der Bereich im Display bleibt solange leer, bis der nächste Titel gespielt wird, 
1.) bitte immer den StreamTitle anzeigen (laut Serial Monitor ist der Titel noch übertragen worden) 
2.) als schlauer Programmierer weißt Du eventuell sogar, wie man den Werbevorspann unterdrücken könnte ...? Ja, zwar nicht im Interesse des Senders, aber ...

Grüße und schöne Feiertage - Alfons.

SerialMonitor: 
13:05:53 AUDIO_info:  SSL has been established in 666 ms, free Heap: 144039 bytes
13:05:53 bitRate:     192Kbit/s
13:05:53 StationName: Radio Oberland
13:05:53 icy-url: ..  http://www.radio-oberland.de
13:05:53 icy-descr:   Unspecified description
13:05:53 AUDIO_info:  MP3Decoder has been initialized, free Heap: 143971 bytes , free stack 2256 DWORDs
13:05:53 lastURL: ..  https://radiooberland--di--nacs-ais-lgc--04--cdn.cast.addradio.de/radioober ... 3ZTU5ZWRlZDFjMmY5MDM
13:05:54 Audiocodec:  MP3
13:05:54 AUDIO_info:  stream ready
**13:05:54 StreamTitle:**
13:05:54 AUDIO_info:  syncword found at pos 0
13:05:54 AUDIO_info:  durationMilliseconds='27096'
13:05:54 AUDIO_info:  MPEG-2.5, Layer I
**13:05:54 StreamTitle: 27096**
13:05:54 AUDIO_info:  Channels: 2
13:05:54 AUDIO_info:  SampleRate: 48000
13:05:54 AUDIO_info:  BitsPerSample: 16
13:05:54 AUDIO_info:  BitRate: 192000
**13:06:11 StreamTitle: PENTATONIX - HALLELUJAH (RADIO EDIT)**
13:06:21 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
13:06:21 AUDIO_info:  syncword found at pos 575
13:06:21 AUDIO_info:  syncword found at pos 0
13:06:21 AUDIO_info:  MPEG-2.5, Layer I
13:06:21 AUDIO_info:  Channels: 2
13:06:21 AUDIO_info:  SampleRate: 48000
13:06:21 AUDIO_info:  BitsPerSample: 16
13:06:21 AUDIO_info:  BitRate: 192000
13:08:55 StreamTitle: TIMBALAND FEAT. ONEREPUBLIC - APOLOGIZE
13:13:56 StreamTitle: ELTON JOHN - STEP INTO CHRISTMAS

schreibfaul1, 2024-12-26:
Hallo Alfons,
ein neuer "Streamtitle" wurde gelöscht bevor die Werbung vorbei ist. Das funktioniert jetzt besser. Ein Überspringen der Werbung ist bei einem Livestream kaum machbar, das ist genauso wie beim Fernsehen. Bei Aufzeichnungen wäre das anders, etwa so wie das der Brave Browser mit der Werbung in Youtube Videos macht,
einen schönen letzten Weihnachtsfeiertag..
vG Wolle

TubaFonse, 2024-12-26:
Hallo Wolle, 
sieht jetzt sehr gut aus - danke für alles und viele Grüße, 
Alfons

closed 2024-12-26 by TubaFonse