schreibfaul1/ESP32-MiniWebRadio #581: bug report (set volume via dlna menu)

clear-sky, 2024-12-24:
Hi Dear Wolle!
thanks for your efforts in increasing ir functions!
tried last update and seems when i try to set volume via DLNA menu volume jumps to maximum value!
please check

clear-sky, 2024-12-25:
Hi Dear wolle!
it seems that valume value in DLNA tab grabs its real value after several switching between radio and DLNA!
after grabing real  volume value switching to DLNA menu disables audio and user need to change the current station

schreibfaul1, 2024-12-27:
Hello clear.sky,
I do not assume that the error only occurs in DLNA mode. The slider was missing the value for the volumeSteps. Hope it works now.
best regards
Wolle

clear-sky, 2024-12-28:
Hi Dear wolle!
thanks for your efforts!
grabbing real value is solved! but as i said disabling audio issue remains!
when i switch from radio to dlna and return again to radio (VIA WEB INTERFACE) audio exists  normally
but when i try to do that exact switch with ir remote audio disappears!

clear-sky, 2024-12-31:
Hi Dear Wole!
please note to this hint:
when i switch from radio to player menu via IR remote (web interface is open): web interface switches to player menu
but when i switch from radio to dlna menu : web interface does not switch to dlna menu ( I only see radio menu with dlna icon in web interface)

TubaFonse, 2024-12-31:
Hi Dear clear-sky and Wolle, 
there is another bug: 
when i switch from radio to player menu via IR remote (web interface is open): web interface switches to player menu. 
But when i further switch **from player menu** via IR remote **to dlna menu**: web interface does not change (remains at player menu)

clear-sky, 2025-01-04:
Hi  Dear Wolle!
Manay many thanks! problem solved!

closed 2025-01-04 by clear-sky