schreibfaul1/ESP32-MiniWebRadio #583: feature request (separate arrow keys from vol+/- and ch+/-)

clear-sky, 2024-12-24:
You may not like this idea, but my experience working with different remotes on audio and video players has proven to me that the volume up and down and channel up and down keys should be separated from the arrow keys.
For example, when setting a timer in this project, the user unknowingly presses the up and down keys, which have different functions in this project and are very annoying.
Because there is a need to change the channel and volume in any part of the menu, good remote control should have separated vol+/-  ch+/- from arrow keys
many many thanks for your efforts

valera39, 2024-12-24:
[clear-sky](https://github.com/clear-sky), I second that, for Android consoles remotes, it's very much needed

Kicadder, 2025-01-03:
Best wishes, and a happy new year to all!
Dear Wolle, thank you very much for sharing this project!
Like Clear_sky and valera39, I would appreciate the separation between arrow keys and Vol+/- and CH+/-.
For Vol+/-, I noticed a latency of several tens of seconds (and other commands too). Could you confirm that you have the same behaviour? Is it something that could improve this? (when willing to reduce volume accidently put to its maximum, to a civilized one, it takes time!)
Thank you!

clear-sky, 2025-01-04:
Hi Dear Wolle!
happy new year- i have an idea:
you can create an option to user to select his/her remote :
option1: small or medium remote(less than 30 keys + long press option)
option2: huge remote (more than 35 keys **without long press option** + separated vol+/- & ch+/-)

schreibfaul1, 2025-01-04:
Hi @Kicadder , hi @clear-sky ,
I could add a long press repeat for keys 26 and 27
[image]
I would leave the arrow buttons unchanged for the "small" remote controls.

clear-sky, 2025-01-04:
Hi Dear Wolle!
is there any limitation to add one more short pressed column?

schreibfaul1, 2025-01-05:
The NEC protocol allows 256 different key codes. I have now added CH+ and CH-. 
[image]
"OK" is the only button that triggers the "fall asleep" function when the button is pressed long.

clear-sky, 2025-01-06:
Hi Dear Wolle!
thank you for adding this two keys!
please note that you just added two more keys but separation is not done!
arrow keys have other jobs BUT if you separate this keys the users that have small/medium remotes will got a trouble!!!
ass i said in 4 posts ago i suggest Separating these two remote owners!
another solution is you create a property for vol+/- and ch+/- keys if user USES THIS KEYS the arrow keys lost their control for vol+/- and ch+/-  BUT They can only navigate through menus!!! 
this way is much more better than first idea and you are skilled enough to do this 
_many many thanks for your efforts_

schreibfaul1, 2025-01-08:
Hallo @clear-sky ,
The idea is that there is still downward compatibility with smaller remote controls with fewer buttons.
When "OK" is pressed, a button or a list entry turns magenta on the display. You can then navigate using the arrow buttons. Otherwise, the arrow buttons are used to change the station or to change the volume.

clear-sky, 2025-01-12:
Hi Dear Wolle!
Thanks for the answer
Typically, when we use the remote, we are far from the device, so it is better to make some modifications in the menu design,
For example, when the user wants to increase or decrease a quantity, it is better to use the up and down keys instead of pressing the OK button.
In general, it is better to design the menu in such a way that the user can work with the device without seeing the screen.