schreibfaul1/ESP32-MiniWebRadio #585: sleep timer buttons

valera39, 2024-12-28:
Hello Dear Wolle!
I'm using three different remotes for the radio that I made, and none of them can change the sleep timer off time in the new firmware. The assigned up and down buttons on the remotes do not change the sleep timer off time. In the last version, I used the left-right buttons to change the sleep timer off time.
My remotes
[image]
Your answer is very important to me.
Success and health in the new year!

schreibfaul1, 2024-12-30:
Hi valera39,
you must mean this layout:
[image]
Arrow right/left changes the buttons and OK is the click.
Here is my ir_buttons.json for the middle remote control in your photo. It works for me.
[ir_buttons.json](https://github.com/user-attachments/files/18276228/ir_buttons.json)
Simply load via the website and perform a reset.

Best wishes for the new year!
Wolle

valera39, 2024-12-30:
Thank you very much, your settings probably make more sense for the control. I will now try to adapt the other two remotes.
I would also like to ask you to make a "reset" button in the web interface of the radio to make it easier to control it.
All the best!

schreibfaul1, 2024-12-30:
Unlike the Toch display, I need to know where I am with the remote control.
I achieve this with "ok" and a button is coloured magenta in the menu.
I can use the right and left arrow keys to jump to the next button. "ok" then activates this button. The same applies to the lists with the up or down arrow.
This already works everywhere except for setting the time for the alarm clock. I haven't managed that yet.
What is the "Reset button" supposed to do?

valera39, 2024-12-30:
The reset button should reboot the radio. similar to the reset button (pin EN) on the ESP32

valera39, 2025-01-02:
Hello Dear Wolle!
is my offer worthy of your consideration?

schreibfaul1, 2025-01-04:
I don't know exactly how useful this is. If the radio runs in an endless loop, a SW reset will not work. Then only a HW reset or briefly switching off the power will help.

closed 2025-01-07 by valera39