schreibfaul1/ESP32-MiniWebRadio #586: Danke und ein gesundes neues Jahr an Wolle und alle User

Arne65, 2024-12-30:
Hallo Wolle ,

ich habe mal eine PCB gemacht hat auf anhieb funktioniert . (V3.6.1)

Display   40MHz
SD-Card 80 MHz

[image]
[image]
[image]

Eventuell inspiriert das den einen oder anderen zu eigenen Projekten .

Falls ihr diese PCB sucht findet ihr das unter :

https://oshwlab.com/kubel/last-mini-web

Auf meiner PCB ist die Beschriftung leider teilweise falsch das war aber mein Fehler im PCB Layout stimmt dann alles .

Ich möchte dir hier  im Namen aller User danken für deine  unermüdliche Arbeit und wünsche allen einen
guten Rutsch .

skassel2024, 2025-01-02:
Hallo Wolle,
Dir und allen Usern ein gutes und gesundes neues Jahr '25. Danke auch von mir für Deine unermüdliche und wunderbare Arbeit.

[Arne65](https://github.com/Arne65)
Spitzenmässig. Kann ich die Platine bei Dir käuflich erwerben? Ich habe Interresse an 3 Stk.

Grüße Sören

schreibfaul1, 2025-01-02:
Hallo Arne,
sieht gut aus, was du entwickelt hast.
Ich könnte das nicht, auch würde ich die SMD Tausendfüßler nicht korrekt einlöten können.
Wenn du magst, werde ich die Bilder und den Link in den Userbereich übernehmen, damit sie besser von Interessenten gefunden werden können.
vG Wolle

Arne65, 2025-01-02:
Hallo Sören ,

ich habe die PCB eigentlich hier reingestellt als Anregung für andere Benutzer.

Die Leiterplatte habe ich entwickelt mit EasyEDA die Software ist vollkommen kostenlos .

Aus EasyEDA ist es mehr als einfach möglich die Herstellung der Leiterplatte bei JLCPCB zu beauftragen .

Entweder nur die PCB oder sogar bestückt fix und fertig.  ( zu fix und fertig habe ich aber mal keinen Plan was die Damen und Herren da aufrufen ich weiß nur das es geht ).

Die Kosten für  5 unbestückte PCB liegen bei etwa 6,64 Euro  also 1,32 Euro pro Stück dazu kommen dann 
Frachtkosten und MwSt bei mir waren das  zusammen 14,12 bedeutet einen PCB unbestückt 2,82 Euro.  Bestellung ,Produktion  bis es in deinem Briefkasten ist etwa 7 Tage. 

Was ich insgesamt als wirklich einen guten Preis nenne :)

Der Preis der PCB wird aber von einigen Faktoren beeinflusst, wie Kupferstärke und einigem mehr das alles kann man aber beim bestellen einstellen . 

Das fertige Projekt findet man unter https://oshwlab.com/kubel/last-mini-web das kann Du dann einfach im Editor öffnen und eventuell an Deine eigenen Bedürfnisse anpassen.

Um aber auf deine Frage zu kommen .

Da die Beschriftung auf meinen 4  noch vorhanden Leiterplatten teilweise falsch und damit irreführend ist würde ich die nur ungerne weiter geben da diese nicht perfekt sind . Das beeinträchtigt zwar nicht die Funktion aber es ist dann eben nicht perfekt .  

Die PCB bei https://oshwlab.com/kubel/last-mini-web sollte die Beschriftung korrekt sein .

Wenn du davon 3 haben willst kann ich dir die gerne überlassen .

Da wie alles  hier vollkommen Finanzinteressenlos ist (und damit noch nochmal einen MEGA Dank an Wolle)

wären das 2,82 pro PCB + Porto denke mal 1,80 in Deutschland  sind dann in Summe 10,26 Euro

Wenn du von einer fertigen PCB redest muss ich nochmal Nachdenken und rechnen .

Und ich müsste wissen welchen ESP32 S3 du drauf haben willst  .

 

VG Arne

Arne65, 2025-01-02:
Hallo Wolle,

na gerne kannst du das in den Userbereich übernehmen .

Das mit dem SMD löten ist nach ein wenig Übung eigentlich total easy . Flussmittel die richtige Lötpaste und Lötspitze ist der Schlüßel zum Glück :)

schreibfaul1, 2025-01-03:
Hallo @Arne65,
habs hierhin kopiert  [self-made devices of the users](https://github.com/schreibfaul1/ESP32-MiniWebRadio/wiki/User-devices)

Du hast Recht, alles ist hier frei von Finanzinteressen. Das Projekt bleibt "open source". Ich hatte bereits mehrere Angebote von Firmen erhalten, auch solche, die PCBs vertreiben.
vG Wolle

Arne65, 2025-01-03:
Hallo Sören 

Top Mega kompakt sieht gut aus .

PayPal ist sicher die beste Wahl.

Ich gebe hier  mal eine E-Mail Adresse dorthin kannst du mir Deinen Postadresse schicken und ich gebe dir dort mein PayPal Konto .

E-Mail für Kontakt : kubelpcb@gmail.com

Das  Du zum Flashen  des ESP einen FTDI Adapter brauchst ist dir bekannt ?
Du musst den ESP über RESET  und BOOT in den Downloadmodus bringen zum Flashen  .
Ich denke das ist auch bekannt ,das geht im Gegensatz zu den fertigen ESP Boards nicht automatisch (wollte es nur bemerken).

Ich habe auf der PCB auf einen USB-UART Transceiver,12Mb/s/1Mb/s, QFN-28 verzichtet da ich glaube darauf haben die wenigsten Lust den auf die PCB zu bringen . 0,5mm Pin-Abstand ist nicht für jeden was . (aus diesem Grund das mit dem FTDI)

Auf der PCB sind die zweit Switch für RESET und BOOT TS-1177-C-B-B  die oder was baugleiches hast du ?
 [][(url](https://www.lcsc.com/datasheet/lcsc_datasheet_2006031232_XKB-Connection-TS-1177-C-B-B_C561512.pdf))

Wenn ich Deine Adresse am WE bekomme schicke ich das dann Montag los .

Schöne Wochenende
vG Arne

Arne65, 2025-01-03:
Hallo Wolle ,

Danke 

Ich kann mir vorstellen das Du da Angebote bekommst.
Deine Arbeit kann eh keiner bezahlen .
Meinen Respekt für deine Arbeit .

Schönes Wochenende

vG Arne

closed 2025-01-23 by Arne65