schreibfaul1/ESP32-MiniWebRadio #587: Web-Interface: A running Playlist cannot be stopped

TubaFonse, 2025-01-02:
Hallo Wolle, 
wenn ich über das Web-Interface eine Playlist starte, kann ich diese nicht mit der Stop-Taste abbrechen: 
[image]
Bei Drücken der Stop-Taste erscheint im Aktivitätsfenster: 
There is no audio file active
Die Playlist läuft aber weiter. 
Am Display des MWR reagiert der Touch-Screen zwar auf Tasten-Klicks, während die Playlist über das Web-Interface gestartet wurde, aber führt nicht zum Abbruch der Playlist. 
Serial Monitor: 
18:43:30 AUDIO_FILE:  AudioCurrentTime 0:34s, AudioFileDuration 2:55s  
18:43:31 click on ..  btn_PL_radio
18:43:40 AUDIO_FILE:  AudioCurrentTime 0:44s, AudioFileDuration 2:55s  
18:43:41 click on ..  btn_PL_fileList
18:43:42 AUDIO_FILE:  AudioCurrentTime 0:46s, AudioFileDuration 2:55s  
18:43:43 click on ..  btn_PL_shuffle
18:43:44 AUDIO_FILE:  AudioCurrentTime 0:48s, AudioFileDuration 2:55s  
18:43:45 click on ..  btn_PL_playAll
18:43:46 AUDIO_FILE:  AudioCurrentTime 0:50s, AudioFileDuration 2:55s  
18:43:47 click on ..  btn_PL_ready
18:43:48 AUDIO_FILE:  AudioCurrentTime 0:52s, AudioFileDuration 2:55s  
18:43:49 click on ..  btn_PL_nextFile
18:43:49 AUDIO_FILE:  AudioCurrentTime 0:53s, AudioFileDuration 2:55s  
18:43:50 click on ..  btn_PL_prevFile

Kannst Du das bestätigen? 
Danke & Grüße, 
Alfons

schreibfaul1, 2025-01-04:
Ja, ich hatte nur die SD Karte abgefragt, aber nicht ob eine Datei über die Cloud abgespielt wird. Das wird jetzt funktionieren.

TubaFonse, 2025-01-04:
Danke Wolle, 
läuft jetzt super! 
LG Alfons

closed 2025-01-04 by TubaFonse