schreibfaul1/ESP32-MiniWebRadio #588: SD, microSD cards and microSD board wiring

Kicadder, 2025-01-04:
Hope this can be of any use...
[SD_microSD_wiring.odt](https://github.com/user-attachments/files/18305060/SD_microSD_wiring.odt)

schreibfaul1, 2025-01-04:
The pins for the SD_MMC are fixed for the ESP32 and freely selectable for the ESP32-S3. With the ESP32, D0 must not have an external PullUP resistor. In the simplest case, the SD card can be connected directly to the pins of the ESP32.

mickeypop, 2025-01-11:
how is it that you leave off the CS on the SD card

CataCluj, 2025-01-26:
> how is it that you leave off the CS on the SD card

It seems it's used in SD_MMC mode that doesn't require CS.
I still couldn't get the SD card to work!

CataCluj, 2025-01-26:
All docs specify 10K Pull-Ups on some pins. These are not the series resistors proposed to be replaced with jumpers.
I'm still confused which pins (apparently even some that are not connected to MCU) need Pull-Ups.
In fact I'm confused because whatever I tried, I couldn't get the SD card to work. The project goes no further than that.
It would be nice to have a mode that uses the built in web browser and IR for all controls and save some stations in flash, without SD card and display (just because it needs SD).

CataCluj, 2025-01-26:
I got mine (MicroSD card) to work. My problem was that I used a MicroSD adapter wired like the SD card.
Problem is, Pin 3 of the Adapter is NOT connected to the MicroSD ground; one has to use Pin 6.
Before I found this I added 10K pullups on all pins, so unsure if all are needed, but it now works for me.

closed 2026-04-18 by github-actions[bot]