schreibfaul1/ESP32-MiniWebRadio #589: feature request (add SPDIF digital output)

Ryszard-wpl, 2025-01-06:
fantastic Radio and software. Thank you very much for this great job.
Is it possible and easy to add SPDIF to radio? I just want to use your radio as RADIO TUNER for rest of my audio system.
I have seen few realisation like for instance https://github.com/Edzelf/ESP32Radio-V2/tree/main
but I am lamer and cannot use part of code into another project......
Thanks in advance.
Ryszard

Kicadder, 2025-01-07:
WM8805 board?

schreibfaul1, 2025-01-07:
If you need an optical output, you can use this module. You can simply connect it to the I2S output of the ESP32.
[i2S to spdif](https://de.aliexpress.com/item/1005007622347810.html?spm=a2g0o.productlist.main.21.32a82889PAkk66&algo_pvid=10ee6c3e-11e7-4d77-a5c9-92a17af40ef3&utparam-url=scene%3Asearch%7Cquery_from%3A)

Ryszard-wpl, 2025-02-20:
I just tried board mention is this request - WM8805 I2S to spdif/Toslink output from Ali..... 
I can hear only silence. Not even one spike or any noise. Output format is default as 24 bit word but my receiver is able to decode it. What to do ?

schreibfaul1, 2025-02-20:
I can't contribute anything about S/PDIF and the WM8805. Maybe there is someone reading here who has experience with this.

Ryszard-wpl, 2025-02-20:
Thank you Wolle! The lack of MCLK is probably the problem. Would it be possible to get it from ESP32 or do I need to insert a 12,288 MHz crystal oscillator?

Ryszard-wpl, 2025-02-21:
Dear Wolle - I found declarations in file common.h
Can I declare one spare PIN for MCLK and I receive this clock to feed this board ?  

        #define I2S_DOUT          25
        #define I2S_BCLK           27
        #define I2S_LRC             26
        #define I2S_MCLK            0  // mostly not used

Ryszard-wpl, 2025-02-21:
Confirm - I declared pin 16 for ESP32S3 as I2S_MCLK in common.h file. It Work! Thank you.