schreibfaul1/ESP32-MiniWebRadio #591: MiniWeb Radio v. 4.0

valera39, 2025-01-08:
Hello Dear Wolle!
Also, hello to all members of the community. How do you look at redesigning the radio a bit? I'm sure your knowledge and skills will enable you to tackle the next peak.
At the moment I have come up with a variation of the radio tab. The station name and the artist of the song, if they don't fit in the allotted field, can be made as a ticker.
You can also display weather forecast, for example, from open-meteo.com, as a ticker. 
[image]
[image]
[image]

I ask everyone who is interested in the development of the best radio in the world to evaluate my proposal.
Thank you!

steve6375, 2025-01-08:
Weather would be good and is something I had looked into.
Ok if easy for end user to set up.

Kicadder, 2025-01-09:
Hello valera and steve,
Very nice design!
I would prefer a direct access to volume control and stations selection (vol+/-, mute, the more frequently used functions on an old fashioned amp/tuner), rather than a VU meter or forecasts.
I am an electronic technician, not a programmer; but isn't it possible to imagine a modular firmware, so that anybody can choose appearance of his main screen before compiling? (maybe defining several zones on the display)
Maybe that would be too complex or too long to program, or too heavy in resources for an ESP32?

schreibfaul1, 2025-01-09:
Hello @valera39,
thank you for your design suggestions. In fact, the current layout is still from the early days and looks a bit old. Internally, the code would have to be simplified and made more readable in some places in order not to lose the overview.
It is probably the end for the small displays with 240x320px (stamp television) and in the future there will only be the TFT displays with 320*480px and the larger HMI displays with 800x480px will be added.

valera39, 2025-01-09:
Dear Wolle!
I completely agree with you, the future is for higher resolution displays. that's why I assumed that my modest contribution is just for the 480x320 display.
If you are interested, I will try to work on the design and functions of the tabs, but I am not good at programming. Perhaps other users will suggest other, more recent ideas

valera39, 2025-01-09:
Once again, I will emphasize that appearance to me is the hallmark, the highlight, of your radio. It was because of its beauty that I paid attention to your project. Initially it seemed complicated to me, it was the first version. But everything worked out from the first time. Now I can say it's the best I've ever built.
Do not stop at what you have achieved, conquer new heights. And numerous users all over the world will help you in your decisions!

clear-sky, 2025-01-11:
Hi all!
weather is nice idea - but not in radio menu-When designing a page professionally, you shouldn't make it too messy .
The only thing that can be added to the radio menu is the encoding type, which can be done using a small icon.
The vumeter can also be designed horizontally, of course, symmetrically, because it is both more beautiful and allows for better use of the limited screen space.

steve6375, 2025-01-11:
The level indicators are not needed or even useful to me - TVs and radios do not have them. A weather indication on the main screen is very useful however ,as is day, date and time. Just my 2p worth :-)

transfinite8, 2025-01-11:
I like a more minimal design. I agree with @clear-sky the weather data should not interfere with the Radio information. 

[image]

With a larger 800x480 display the weather data could go in its own grid outside of the Radio grid.

My number one request for Version 4 would be support for capacitive touch displays. But no matter what direction Wolle decides to pursue going forward, I'm a big fan of MiniWeb Radio!

valera39, 2025-01-12:
For example, the weather ticker in the Yo-radio project does not interfere with the perception at all. As for the arrow level indicators, I repeat that this is just a part of the visual design of the large display. And don't compare this radio with a TV or just a receiver, this project has gone far beyond just internet radio, it is a multimedia device. And since it is, the visual design of the display plus a homemade beautiful case will give a head start to many industrial devices.
Suggest your ideas, help the esteemed Wolle

clear-sky, 2025-01-12:
Menu design is also very difficult to do professionally.
Mr. Wolle has considered two user interfaces for the radio, one is the resistive touch and the other is the remote control, which has made menu design very complicated.
Resistive touch is not a pleasant user interface. If I were Wolle, I would remove this user interface.

Bigpriap, 2025-01-12:
For me it is the most interesting and beautiful project at the moment. Thanks to the author!!! Three things would lead him to perfection:
1) the possibility of receiving FM radio stations, with cheap modules that have bookstores: preferably TEF6686, but the other options are also ok: si4703, tea5767.
2) the ability to read audio files from usbflash drive, either natively for ESP23 S3 or with an external module such as CH376
3) spdif output to connect to other devices

valera39, 2025-01-12:
Dear [clear-sky](https://github.com/clear-sky), I'd have to agree with you about the menu button. But on the touchscreen exclusion, I strongly disagree with you. 
First of all, for a large screen it is the most suitable control option. Secondly, you should not transfer all control functions to the remote control only. Users have different remotes with different number of buttons, so the combination of keys and commands will have to be complicated. It's also necessary to have a backup control channel, and with a display like this, it would be foolish not to use it.
And note, with many factory multimedia centers, some functions can only be performed with buttons or switches.
При отсутствии сенсорного экрана можно применить энкодер и перемещаться по меню с его помощью.
Кстати, как вам такое меню?
[image]

It is also possible to make several submenus to the main menu

Bigpriap, 2025-01-12:
Maybe an optional encoder control will satisfy everyone?

clear-sky, 2025-01-12:
Hi Dear Walera39!
Thanks for the answer!
I understand that you disagree with me on the touch interface.
The reason for my suggestion is that resistive touch is not as pleasant as capacitive touch. but Removing this interface would eliminate a lot of complexity.
Intimate chat:
As far as I know, the Russians have a strange interest in rotary encoders, but I don't know why!!!

clear-sky, 2025-01-12:
I agree with rotary encoder too!

valera39, 2025-01-12:
Forgot to finish the bluetooth icon :)

valera39, 2025-01-12:
Hi Dear [clear-sky](https://github.com/clear-sky)!
I don't think that the love for encoders and other knobs that turn only Russians have:)
This is more likely to apply to radio amateurs, especially the older ones, who once built their first radio receiver. And this is the feeling when you slowly turn the knob and through the crackle and noise of the airwaves pops your favorite music or speech of the announcer. It is difficult to explain in words, and even more so with the help of an online translator. It is nostalgia for old tube receivers with their unique design and warmth.
And of course, the “mysterious Russian soul” :)
You've probably heard this expression.

clear-sky, 2025-01-12:
Hi Dear Valera!
My knowledge of Russians comes from movies. I don't find Russians mysterious, but they keep secrets well.
People in most of the world, regardless of geography, are kind, simple, and friendly, and I love this group of people.

Kicadder, 2025-01-13:
I think it is difficult to define what can be displayed on a screen without knowing its definition. It would be a shame to display the same informations in 320 x 480 and in 800 x 480, by simply enlarging! 320 x 480 means 153k pixels. 800 x 480 means 384k pixels. The question therefore arises as to the direction that the MiniWeb Radio will take in the foreseeable future: a small, inexpensive but powerful device, or a more substantial, powerful device, but a little more expensive (for the moment, but prices will probably drop!).
Will version V4.0 be in the same spirit as currently, or will there be a clear change in the display, HMI (rotary encoders)... and the code? (Keeping in mind that an ESP32 has appreciable power... but not unlimited).
And last question, will Wolle want to embark on an adventure that could become difficult, or stay in better mastered, known and therefore more predictable and manageable programming?
That is the question ;-)

clear-sky, 2025-01-13:
Hi Dear Kicadder!
ESP32-P4 can play 1080 video! thus handling a 800x480 display can be done easyly
but this is possible wolle leaves this project and start  a mini web TV project!!!
But if we want to do things wisely, this screen is quite enough for a radio, but designing a professional menu will be a very difficult task.

valera39, 2025-01-13:
I wonder if the esteemed Wolle reads our arguments about the future of radio?

schreibfaul1, 2025-01-13:
Thanks for all the suggestions. I like the layout with the menu. I think there will soon be a larger display (800x480) with the S3. There are already several manufacturers offering such a display. The connections of the RGB HMI display are already connected to the S3 (16 pins for the colour + VSYNC, HSYNC, PCLK, DE).
Advantage: little effort and tinkering, fast graphics, good resolution. No further drivers are required, Espressif provides special routines and the graphics are done by LVGL.
Disadvantage: 20 GPIOs are required for the display + 2 (SDA, SCL) for the touchpad. Only a few pins remain free.
There are displays with built-in audio, albeit mono. However, it is possible to connect a second DAC in parallel, i.e. to connect the BCLK, SCLK, DOUT connections to an external DAC.

valera39, 2025-01-13:
Hello Dear Wolle!
I'm glad you like my design suggestions. Could you please provide a link to the 800x400 resolution display you plan to apply to your project?

clear-sky, 2025-01-13:
Hi Dear wolle!
It is not wise to limit the radio capabilities due to the use of a larger screen. It is better to wait until the P4 comes out so that we do not have any limitation because of lack in GPIO
_MWR ver4 powered by esp32-p4_

valera39, 2025-01-13:
Hi Dear [clear-sky](https://github.com/clear-sky)!
The esp32-p4 is already available, I've already provided links where you can buy it. But the problem is that not every user can afford to buy it at this price. There may be other restrictions, like shipping. But you can't do without a quality display in any case. And you can buy it more easily and inexpensively.

valera39, 2025-01-13:
Hi Dear Wolle!, will this one work? https://aliexpress.ru/item/1005006715818797.html?sku_id=12000038078760735&spm=a2g2w.productlist.search_results.1.4d8945e5HwJsPb

valera39, 2025-01-13:
Hello Dear Wolle! 
You mean this? https://aliexpress.ru/item/1005007794596450.html?sku_id=12000042227683917

clear-sky, 2025-01-15:
Hi Dear Valera!
I agree with you BUT
Gradually, as demand increases in the market, the price of the board decreases.
Meanwhile, major manufacturers such as espressif and AI Tinker have not yet launched their own modules on the market.
it is because of the immaturity of this module.
If you see that Mr. Wolle has not yet moved towards this microcontroller, because The P4 is more like a processor than a microcontroller, and it doesn't even have the ability to connect directly to the network. Some companies connect this processor to the Internet with an Ethernet module. Others add Wi-Fi connectivity with the help of an ESP32C6.
 Its high price is also the next reason.

schreibfaul1, 2025-01-15:
Hello valera,
if it is the display: https://www.waveshare.com/esp32-s3-touch-lcd-4.3b.htm it will work. There are more and more manufacturers of ESP32-S3 HMI displays, and prices will fall. These displays are controlled via horizontal and vertical synchronisation, just like old televisions bulbs. There is therefore no longer a special driver.

I am in the testing phase, and am able to display some buttons and sliders. My idea is to use the LVGL UI Studio for graphic elements. This will avoid a lot of programming work. And you can design your own layouts without any programming knowledge.

clear-sky, 2025-01-15:
Hi Dear Wolle!
https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3B/ESP32-S3-Touch-LCD-4.3B-Sch.pdf
In terms of hardware, this display has only one advantage, and that is that it is an IPS display.
If you use this display, it is not practical for the S3 due to the use of many IOs.

If you care about suggestions, I suggest two other options.
1-https://nextion.tech/datasheets/nx8048t070/
This display has a separate processor and does not require a driver. It uses a serial protocol to connect to the processor, and HMI programming is easy to implement and has extensive documentation.
https://github.com/itead/ITEADLIB_Arduino_Nextion
https://wiki.iteadstudio.com/Nextion_Editor_Quick_Start_Guide
The most important disadvantage of this display is its relatively high price.
2-FT811 from brtchip
https://brtchip.com/product/ft811/
https://github.com/RudolphRiedel/FT800-FT813
This one has only one disadvantage  , and that is that it can be a little tricky to set up.
Benefits include:
The communication protocol with the processor is SPI. this means only 4 IO will be used
The user can choose from a wide range of displays according to their taste and budget.
The capacitive touch interface connects to the chip via an I2C interface, not to the processor.
Capacitive touch is also selected and determined by the user.
Due to the use of discrete components, it is possible for users to purchase its parts from different shopping centers.

valera39, 2025-01-15:
Hello Dear Wolle!
It makes sense to buy the display you recommend https://www.waveshare.com/esp32-s3-touch-lcd-4.3b.htm
Or should we wait for other options. If I understand correctly, your next radio will be on this particular display?
I wouldn't want the display I bought to be unnecessary.

valera39, 2025-01-15:
Hi Dear [clear-sky](https://github.com/clear-sky)!
Do you have a link from any store to the second display you suggested?

schreibfaul1, 2025-01-16:
Hi, it's really just about using a display on which more details can be shown. An HMI display seems to be a good solution, especially as there are already several manufacturers. The TFT displays with 480x320px remain as before. However, 320x240px seems to be too small.
It is still too early to buy a display.

clear-sky, 2025-01-16:
Hi Dear Valera!
you can find lots of stores here!
https://github.com/RudolphRiedel/FT800-FT813
for example:https://www.hotmcu.com/5-graphical-lcd-capacitive-touch-screen-800x480-spi-ft811-p-301.html

clear-sky, 2025-01-16:
Hi Dear Wolle!
thanks for the answer-It would be great if you could continue to support current displays.
The display you have selected is essentially a raw display.By raw display, I mean a display that does not have an interface chip, such as ILI9488.
These displays are divided into two categories:
1-50 pin standard:The size of this category of displays starts at 7 inches. the resolution is usually 800x480(AT070TN92) or 1024x600(HT090DQB)
2-40 pin standard:The size of this category of displays starts at 4.3 inches and usually continues to less than 7 inches.The resolution of these is usually 480x272(for example AT050TN33), but newer models, especially IPS models, support 800x480 resolution. 
The 50-pin standard can be converted to a 40-pin standard with the help of an adapter.
If any user of this display wants to use it directly without a driver, they will have to use some kind of voltage booster to power the backlight.
Capacitive touch has no connection to the display and is connected directly to the processor.(for example gt911)
**_Advantages of raw displays:_**
No interface chip is needed.
The user can purchase just the display without purchasing the driver and connect it to his processor using the FPC socket
The HMI of this type of display is LVGL, which is very advanced and has good support.
**_Disadvantages:_**
The number of IOs used is high.
All HMI processing is done by the processor.
Final result:
Raw displays are ideal for the P4, and S3 and esp32 users should say goodbye!!!

schreibfaul1, 2025-01-17:
HMI displays have these connections:

[image]

Theoretically there are 8 pins for red, 8 pins for green and 8 pins for blue. 
We use 16 bits as color depth. 
This is RGB565, so 5 bits for red, 6 for green and 5 for blue.
Of the possible 49 GPIOs of the S3, we cannot use GPIO 22...37 (internal for FLASH and PSRAM).
We need 16 GPIOs for color, 4 for synchronization. 
If the ESP32-S3 is on the display, you save the wiring.

valera39, 2025-01-17:
Hello Dear Wolle!
Am I understanding correctly that you are experiencing this display https://www.waveshare.com/4.3inch-capacitive-touch-lcd.htm

clear-sky, 2025-01-18:
Hi! Dear Wolle!
found this in reddit!!!

[image]

take a look here:
https://seengreat.com/product/274/e-paper-display-driver-board

schreibfaul1, 2025-01-18:
E-ink displays are energy-saving but also extremely slow.
The display that valera has found is usable if you want to wire everything yourself.

RGB HMI displays are fast and reproduce colours better, even if it doesn't look like that in the photo.

[image]

valera39, 2025-01-19:
Hello Dear Wolle!
You can add small icons of the flag of the country where the radio is currently playing. The data can be taken from the playlist, where the country of the radio station is specified, and place the flag collection on the map. 
Maybe this is possible already in version 3.6?

[image]

And in the future, on large displays, it will be possible to make a separate information tab, where you can place the parameters of the playing radio station and other data that are not relevant on the main screen

clear-sky, 2025-01-19:
Hi Dear Wolle!
thanks for the answer-
have you any plan to use rgb display in 888 mode ?( i mean in p4)

schreibfaul1, 2025-01-20:
Hi clear-sky,
for RGB888 we need 24 pins, that would really only be something for the P4.
So far we have RGB565 and this will initially be the case with the HMI display, although RGB888 would be possible with HMI displays.

Jason2866, 2025-01-26:
Sharing some experiences we made with project Tasmota and displays. An S3 is fine driving displays up to 480x480. Using 800x480 is on the edge what the S3 can when other tasks should run well too. Currently the display support for RGB displays like the waveshare you mentioned earlier is broken since IDF 5.3 is under the Arduino hood. Before when IDF 5.1 was used (Arduino Core 3.0.x) it worked well. Broken means we have horrible jitter which makes the use of the display nearly impossible. We are still searching (and struggling) what change in IDF has changed (decreased) the performance. There was a heavy rewrite in the DMA part.
Probably it is now a little bit slower "somewhere" and we get in trouble with providing the data in time. As said before smaller sized displays do still work fine.

valera39, 2025-04-10:
Hello Dear Wolle!
Can you tell us about your plans for the fourth version of your radio?
Or do you have no more interest to change and add something and develop the project in general?
The radio has got a big display, maybe there is a sense to radically change some elements and appearance of the main screen, bigger and more informative level indicator, make a page menu. 
I think numerous fans of your project all over the world who follow your work will be interested.
 For example, I would certainly express my gratitude to you for such a wonderful project in the form of monetary incentives, as a gift, but unfortunately because of political strife I will not be able to do it.
With respect and all the best!

schreibfaul1, 2025-04-10:
There are still the small displays. The graphics should be the same for everyone.

I have added an on-screen keyboard. If you have entered the SSID/PW in the code (in common.h), nothing changes. Otherwise the `WiFi Settings` will open (or with `pwd` in the terminal) This replaces the previous access point. And the networks.csv no longer exists.
[image]
If everything goes well, the V4 will be the master at Easter.

valera39, 2025-07-27:
It probably doesn't interest anyone but me.

schreibfaul1, 2025-07-28:
Hi valera,
I apologise for not replying. Most of the work recently has been on the AudioI2S library.
You can now connect 48KHz BT transmitters and DAcs that don't work with other sample rates.
If you have an old ESP32 you can use it as a Bluetooth DAC.
The OPUS decoder can now play the hybrid mode.
Internally I now use unique pointers to avoid memory leaks.
For the MWR I am only doing bug fixes. It won't continue until the autumn. We have a large property with a lot of work in the summer and I'm on holiday in August.

Best regards
Wolle

valera39, 2025-07-28:
Thank you for taking the time to respond, I will be waiting for changes not only in the playback decoders, but also in the appearance. Unfortunately, I did not understand if there are plans to change the appearance of the shell.

valera39, 2025-08-02:
I see you're responding in other threads. I hope that you will pay attention to my suggestion, even if it turns out to be unacceptable, please let me know.

valera39, 2025-08-02:
It's just that sometimes, in addition to questions, I have ideas on how to make the radio even better. They may not always be the most successful, but I hope some of them will be useful to you.

clear-sky, 2025-08-05:
Hi all!
esp32-p4 has not yet reached its final maturity. A Chinese manufacturer called wireless-tag has produced this module at a reasonable price, but it lacks wireless.(WT0132P4-A1)

Jason2866, 2025-08-06:
There are already P4 boards with big display for ~33€ on Ali.
And as already said Wifi is really easy to use. After a few changes the project Tasmota does run stable. The changes we have to do to support the display are more and not so easy and quick done.