schreibfaul1/ESP32-MiniWebRadio #595: No Issues PCB Part 2

Arne65, 2025-01-11:
Hi everyone,

I have developed a second PCB.

It has the dimensions of the display 480x320.

Will publish it soon like the first one on oswhlab.

So whoever feels like it.

[image]
[image]
[image]

alfrede, 2025-01-13:
Hi Arne, hättest du ein Schaltbild von der Platine? Ich würde gerne auf der Basis eine PCB mit THT Widerständen erstellen.
Natürlich nur wenn es für Dich okay wäre.

Arne65, 2025-01-13:
Hallo ,

klar kannst du das machen 

[Schematic_MINI-WEB-RADIO-Smal_2025-01-13.pdf](https://github.com/user-attachments/files/18394762/Schematic_MINI-WEB-RADIO-Smal_2025-01-13.pdf)

closed 2025-01-23 by Arne65