schreibfaul1/ESP32-MiniWebRadio #596: Attempt to install version 3.6

samand587, 2025-01-12:
I get errors during compilation. What needs to be fixed?

[image]

Arne65, 2025-01-12:
You don't seem to have any free space on your hard drive

clear-sky, 2025-01-12:
i am wondering why .platformio is so huge?
in my pc .platformio is 11G !!!

samand587, 2025-01-13:
Hi! I installed version 3.6 and am surprised! Screen No. 2 turned out to be like in version 2. When you press any key, screen No. 3 opens at the bottom of screen No. 2. Pressing any key at the bottom of screen No. 3 opens screen No. 1 (initial).  Why did this happen? I'm in shock!

skassel2024, 2025-01-13:
Hi [samand587](https://github.com/samand587), try an "Erase Flash"

samand587, 2025-01-13:
I'm a newbie! How to do it? Thanks!

skassel2024, 2025-01-13:
[image]

schreibfaul1, 2025-01-13:
It could be that there is no space on the "C:\" drive.
[image]

samand587, 2025-01-14:
Hi! Clearing the memory does not help! All the problems remain!

samand587, 2025-01-14:
Hi!  I did everything from the very beginning. Nothing changes, everything is the same on the screen as described above. What happened to version 3.6?

skassel2024, 2025-01-14:
try to delete C:\Users\..........\.platformio\packages. then reinstall platformio in vscode. In platformio.ini select the correct board and board_build.partition. Now you should restart vscode, perhaps twice

samand587, 2025-01-15:
Hi! Reinstalled Visual VS Code, erased all the old data, and updated everything on the SD card. But the result is exactly the same as described above. What's with the project, dear author?

schreibfaul1, 2025-01-15:
In the new version, almost all functions can be controlled with the remote control, including scrolling through lists.

I have no real explanation for this. The most radical method is to delete (or rename) the entire folder on the hard drive and clone the project again. This will take some time, but PlatformIO will reload all the data and add all the required libraries. Alternatively, you can search for the .platformio folder and delete the 'packages' and 'platforms' folders. These will then be recreated automatically.

samand587, 2025-01-15:
Nothing is working! How do I install version 3.4?

hevet, 2025-01-15:
https://github.com/schreibfaul1/ESP32-MiniWebRadio/archive/refs/tags/V3.4.zip

samand587, 2025-01-16:
Hi! When installing 3.4, this error is displayed.  What should I watch?

[image]

samand587, 2025-01-16:
These are the notifications I receive when I start working.

[image]
[image]

schreibfaul1, 2025-01-17:
You constantly have other errors, why does the Git repository cloning not work?

[image]

Then PlatformIO will do everything for you, download the correct Arduino version, the appropriate IDF version and all the necessary libraries.

samand587, 2025-01-17:
You misunderstood me!  Version 3.6 compiles without problems, but it doesn't want to work. I wrote about the problems above. The last picture with errors is for version 3.4. I tried to compile version 3.4 from the archive.

samand587, 2025-01-19:
Version 3.6 is installed! Thanks to the author!

closed 2025-01-19 by samand587