schreibfaul1/ESP32-MiniWebRadio #598: Spotify Integration?

Arne65, 2025-01-16:
Hello Wolle

Can you imagine integrating Spotify?

https://github.com/FinianLandes/SpotifyEsp32
Wath do you think about that ?

Best regards

schreibfaul1, 2025-01-17:
Hello Arne,
As I see it, the Spotify app must be installed somewhere in the home network.
I don't think you can access Spotify directly because of the encryption and the subscription

Arne65, 2025-01-17:
Hello Wolle,

I think Spotify doesn't have to run as an app on the home network. The whole thing works via a web API. I have that in my io broker and it works wonderfully. I'll test it when I get the chance with the example https://github.com/FinianLandes/SpotifyEsp32. There's a library SpotifyEsp32.h and I'll look into it. For me personally, connectivity would be more important than a flickering mouse cinema (VU meter). This isn't meant as criticism, I'm just not that into it. But everyone can see it however they want. :)

To integrate weather data, you would also have to use an API to obtain geographically correct data.

Thank you very much for your quick answer.

schreibfaul1, 2025-01-17:
I have no idea about Spotify, but I can imagine that the web API must be installed somewhere. Spotify will not allow direct access from the ESP32.

Arne65, 2025-01-17:
You are of course absolutely right that Spotify has to be available on a device in order to retrieve the necessary information with the ESP and then control the player. I will first try to connect to Spotify with an ESP.
This also works with the io-broker. Here I have a Java script. The player is then controlled via a virtualized interface or via an installed NS panel, which I flashed with lovlesui

then it looks like this

[image]

closed 2025-01-23 by Arne65