schreibfaul1/ESP32-MiniWebRadio #600: Remote control in version 3.6

samand587, 2025-01-19:
I tried to connect the remote control. Commands are not accepted or configured in the browser. The remote looks like this. The sensor of the remote control is connected correctly.

[image]

Kicadder, 2025-01-20:
Hello samand,
You could check your RC (remote control) with a camera (that of your phone for example). The RC, when emitting (pushing a key) sends a IR ray that is visible by the camera.
If this is OK, you could check the IR receiver IC : +5V, ground and output. If you have a multimeter, the voltage at output must vary of several tenth of volts when you press a key on the RC.
I have already used the same RC you use : no problem with that model

samand587, 2025-01-20:
Thanks for the tips! I've checked it all, and everything is fine. And what address do you specify for the remote control?

schreibfaul1, 2025-01-20:
The address code is fixed for each remote control. The "IR command" field is updated each time a button is pressed. If everything is OK, this will also happen. You can then assign the button to a function. This is necessary so that as many different remote controls as possible can be used

samand587, 2025-01-20:
How can I find out the fixed code for my remote?

samand587, 2025-01-21:
Hello, author!  I changed two remote control sensors- the commands do not go through the browser.  When you press a button on the remote, the sensor reacts, but there are no changes in the browser. I've tried different browsers. The sensor is standard from Arduino, connected to pin 35.

skassel2024, 2025-01-21:
Hello,[samand587](https://github.com/samand587)
Standard sensor from Arduino are very poor. Please read closed Issue #563.

samand587, 2025-01-25:
TSOP34838 It works great!

closed 2025-01-25 by samand587