schreibfaul1/ESP32-MiniWebRadio #601: Radio goes into reboot

valera39, 2025-01-19:
Hello Dear Wolle!
When switching to this station https://instrumental-orchestral-hits.stream.laut.fm/instrumental-orchestral-hits the radio goes into reboot and so on ad infinitum
COM port log:
ESP32 Chip: ESP32-S3
Arduino Version: 3.1.0
ESP-IDF Version: 5.3.2
MiniWebRadio     Version 3.6.1g  - Jan 08/2
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8371924 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 7573.6 MB by 7595.0 MB free
[   316][W][main.cpp:439] defaultsettings(): 0x001A,  ZERO
[   322][W][main.cpp:439] defaultsettings(): 0x0013,  ONE
[   329][W][main.cpp:439] defaultsettings(): 0x001B,  TWO
[   335][W][main.cpp:439] defaultsettings(): 0x0007,  THREE
[   341][W][main.cpp:439] defaultsettings(): 0x0017,  FOUR
[   347][W][main.cpp:439] defaultsettings(): 0x001F,  FIVE
[   353][W][main.cpp:439] defaultsettings(): 0x0003,  SIX
[   360][W][main.cpp:439] defaultsettings(): 0x0016,  SEVEN
[   366][W][main.cpp:439] defaultsettings(): 0x001E,  EIGHT
[   372][W][main.cpp:439] defaultsettings(): 0x0002,  NINE
[   378][W][main.cpp:439] defaultsettings(): 0x0043,  MUTE
[   385][W][main.cpp:439] defaultsettings(): 0x0059,  ARROW RIGHT
[   391][W][main.cpp:439] defaultsettings(): 0x0055,  ARROW LEFT
[   398][W][main.cpp:439] defaultsettings(): 0x004D,  ARROW DOWN
[   405][W][main.cpp:439] defaultsettings(): 0x005E,  ARROW UP
[   411][W][main.cpp:439] defaultsettings(): 0x004C,  MODE
[   418][W][main.cpp:439] defaultsettings(): 0x005D,  OK
[   424][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  -
[   430][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  PAUSE/RESUME
[   437][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  STOP
[   444][W][main.cpp:439] defaultsettings(): 0x0053,  ON/OFF
[   450][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  RADIO
[   457][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  PLAYER
[   464][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  DLNA
[   470][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  CLOCK
[   477][W][main.cpp:439] defaultsettings(): 0x0044,  OFF TIMER
[   483][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  VOLUME+
[   490][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  VOLUME-
[   497][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  << 30s
[   504][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  >> 30s
[   511][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  SLEEP
[   517][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  CANCEL
[   524][W][main.cpp:439] defaultsettings(): 0xFFFFFFFF,  -
[   530][W][main.cpp:443] defaultsettings(): 0x0001,  IR address
00:00:00 setup: ....  seek for WiFi networks
00:00:00 WiFI_info:   add SSID: mySSID
00:00:00 WiFI_info:   add SSID: Home_2487201_RPT_2
00:00:00 WiFI_info:   add SSID: Home_2487201
00:00:00 WiFI_info:   add SSID: Home_2487201_RPT
[  4571][W][STA.cpp:541] disconnect(): STA already disconnected.
00:00:00 WiFI_info:   Connecting WiFi...
00:00:00 WiFI_info:   mDNS name: MiniWebRadio
00:00:00 setup: ....  connected to Home_2487201_RPT_2, IP address is 192.168.1.73, Received Signal Strength -48 dB
00:00:00 ftpServer:   Buffers allocated: 4622 bytes
00:00:00 rtime_info:  Initializing SNTP
00:00:00 AUDIO_info:  commFMT = Philips
00:00:00 setup: ....  number of saved stations: 109
00:00:00 setup: ....  number of saved favourites: 109
00:00:00 setup: ....  current station number: 41
00:00:00 setup: ....  current volume: 14
00:00:00 setup: ....  volume steps: 100
00:00:00 setup: ....  volume after alarm: 10
00:00:00 setup: ....  last connected host: http://163.172.77.142:8006/;stream/1
00:00:00 setup: ....  connection timeout: 2500 ms
00:00:00 setup: ....  connection timeout SSL: 3500 ms
00:00:00 action: ...  switch to station 41
00:00:00 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
00:00:00 AUDIO_info:  buffers freed, free Heap: 174655 bytes
00:00:00 AUDIO_info:  connect to: "163.172.77.142" on port 8006 path "/;stream/1"
00:00:00 AUDIO_info:  Connection has been established in 54 ms, free Heap: 174103 bytes
00:00:00 setup: ....  Ambient Light Sensor BH1750 found
00:00:00 StationName: %title - %artist
00:00:00 bitRate:     96Kbit/s
00:00:00 icy-url: ..  https://xocotakamu.jimdo.com/
00:00:00 AUDIO_info:  MP3Decoder has been initialized, free Heap: 171947 bytes , free stack 5600 DWORDs
00:00:00 lastURL: ..  http://163.172.77.142:8006/;stream/1
00:00:00 Audiocodec:  MP3
00:00:00 AUDIO_info:  stream ready
00:00:00 ftpServer:   Ftp server waiting for connection on port 21
00:00:00 AUDIO_info:  syncword found at pos 0
00:00:00 BT-Emitter:  KCX_BT_Emitter found 
00:00:00 StreamTitle: Juan Torres Y Organo Melodico - El Sube y Baja
00:00:00 AUDIO_info:  MPEG-2.5, Layer I
00:00:00 AUDIO_info:  Channels: 2
00:00:00 AUDIO_info:  SampleRate: 44100
00:00:00 AUDIO_info:  BitsPerSample: 16
00:00:00 AUDIO_info:  BitRate: 96000
00:00:00 BT-Emitter:  POWER ON 
00:00:00 BT-Emitter:  Status -> Disconnected
00:00:00 BT-Emitter:  Version KCX_BT_RTX_V1.4
00:00:00 BT-Emitter:  Volume -> 16
00:00:00 BT-Emitter:  OK+POWEROFF_MODE 
00:04:50 BT-Emitter:  Status -> Scan...
00:04:53 BT-Emitter:  Status -> Connected
00:04:54 BT-Emitter:  no response: last command was: AT+POWER_OFF
[ 13534][E][DLNAClient.cpp:230] readHttpHeader(): contentlength is not given
00:04:54 DLNA_server: [0] 192.168.1.4:6000  dlink-E247EB
00:04:54 DLNA_server: 1 media server found
00:05:00 BT-Emitter:  no response: last command was: AT+BT_MODE?
00:05:03 Webpage:     index.html
00:05:04 webSrv: ...  GetFolder "/"
00:05:04 webSrv: ...  Load from SD  "/stations.json"
00:05:04 webSrv: ...  Load from SD  "/ir_buttons.json"
00:05:06 BT-Emitter:  no response: last command was: AT+VOL=16
00:05:17 action: ...  switch to station 42
00:05:17 AUDIO_info:  buffers freed, free Heap: 170099 bytes
00:05:17 AUDIO_info:  connect to: "instrumental-orchestral-hits.stream.laut.fm" on port 443 path "/instrumental-orchestral-hits"
00:05:18 AUDIO_info:  SSL has been established in 904 ms, free Heap: 169959 bytes
00:05:18 bitRate:     128Kbit/s
00:05:18 StationName: Instrumental Orchestral Hits
00:05:18 icy-url: ..  http://laut.fm/instrumental-orchestral-hits
00:05:18 icy-descr:   Unspecified description
00:05:18 AUDIO_info:  MP3Decoder has been initialized, free Heap: 169719 bytes , free stack 3216 DWORDs
00:05:18 lastURL: ..  https://instrumental-orchestral-hits.stream.laut.fm/instrumental-orchestral-hits
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40380f86  PS      : 0x00050033  A0      : 0x40380e87  A1      : 0x3fccd810
A2      : 0xfffffec0  A3      : 0x00000001  A4      : 0x00000034  A5      : 0x3fcadf54  
A6      : 0x00000002  A7      : 0x60023000  A8      : 0x00000000  A9      : 0x3fca1000
A10     : 0x3fcadf20  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000002  A15     : 0x00000004  SAR     : 0x0000000e  EXCCAUSE: 0x0000001d
EXCVADDR: 0xfffffec0  LBEG    : 0x40055871  LEND    : 0x40055882  LCOUNT  : 0xffffffea  

Backtrace: 0x40380f83:0x3fccd810 0x40380e84:0x3fccd820 0x0005001e:0x3fca6874 |<-CORRUPTED

ELF file SHA256: dc960a997

Rebooting...

Please check this link for yourself.
Thank you!

valera39, 2025-01-22:
Dear, Wolle is there any hope that you will pay attention to my question?

valera39, 2025-01-22:
Dear, Wolle!
I apologize for my persistence, but my question was asked more than three days ago, but there is no answer. Although I see that you are active in the thread and answering other questions asked much earlier. Does this mean that my question is stupid enough to be ignored or is there some other reason?

schreibfaul1, 2025-01-22:
It is difficult to find the cause, especially as the error log unfortunately says nothing except a memory violation. I found a pointer in the audio library that was moved before the memory was released. It could have been the cause, but I'm not sure. Please reload the libraries.

valera39, 2025-01-22:
Thank you very much for the reply, I will try to update the libraries.

valera39, 2025-01-23:
Hello Dear Wolle!
I have updated the libraries, but that didn't do the trick. Interesting feature: if I paste a link to the radio in the web interface of miniweb radio. the radio works fine. But as soon as I add this link to a playlist, the radio starts rebooting when I switch to this station.
In other internet radio projects this station plays perfectly when added to the playlist.
Please check, add this station to your playlist and try to play it while switching between stations
https://instrumental-orchestral-hits.stream.laut.fm/instrumental-orchestral-hits
I will be glad if other community members can help me as well.
I look forward to hearing what you have to say!

valera39, 2025-01-23:
I think I found the solution, I shortened the link to https://instrumental.stream.laut.fm/instrumental-orchestral-hits.
Now everything is working fine. I apologize for the inconvenience, I consider the issue closed.

valera39, 2025-01-23:
Experimenting, I noticed the following: if you shorten the link to https://instrumental.stream.laut.fm/instrumental-orchestral-hits and write this variant directly to the SD card by removing it, when you turn on the radio, this radio station plays normally and you can switch other stations with the remote or touchscreen.
But when I open the web page with this station playing, I immediately reboot.
There's probably some connection to the web interface. I used different browsers.
I hope my experiments will help in solving the problem

valera39, 2025-01-23:
Here's the monitor log from the moment I turned on and then opened the Chrome browser.
Playback is not restored after a reboot. In Firefox browser there is a cyclic reboot.
Is there anything else I can do to help?
ESP32 Chip: ESP32-S3
Arduino Version: 3.1.0
ESP-IDF Version: 5.3.2
MiniWebRadio Version 3.6.1i  - Jan 17/2025
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8371916 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 7573.6 MB by 7595.0 MB free
[   314][W][main.cpp:445] defaultsettings(): 0x001A,  ZERO
[   321][W][main.cpp:445] defaultsettings(): 0x0013,  ONE
[   327][W][main.cpp:445] defaultsettings(): 0x001B,  TWO
[   333][W][main.cpp:445] defaultsettings(): 0x0007,  THREE
[   339][W][main.cpp:445] defaultsettings(): 0x0017,  FOUR
[   346][W][main.cpp:445] defaultsettings(): 0x001F,  FIVE
[   352][W][main.cpp:445] defaultsettings(): 0x0003,  SIX
[   358][W][main.cpp:445] defaultsettings(): 0x0016,  SEVEN
[   364][W][main.cpp:445] defaultsettings(): 0x001E,  EIGHT
[   371][W][main.cpp:445] defaultsettings(): 0x0002,  NINE
[   377][W][main.cpp:445] defaultsettings(): 0x0043,  MUTE
[   383][W][main.cpp:445] defaultsettings(): 0x0059,  ARROW RIGHT
[   390][W][main.cpp:445] defaultsettings(): 0x0055,  ARROW LEFT
[   396][W][main.cpp:445] defaultsettings(): 0x004D,  ARROW DOWN
[   403][W][main.cpp:445] defaultsettings(): 0x005E,  ARROW UP
[   410][W][main.cpp:445] defaultsettings(): 0x004C,  MODE
[   416][W][main.cpp:445] defaultsettings(): 0x005D,  OK
[   422][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  -
[   428][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  PAUSE/RESUME
[   436][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  STOP
[   442][W][main.cpp:445] defaultsettings(): 0x0053,  ON/OFF
[   449][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  RADIO
[   455][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  PLAYER
[   462][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  DLNA
[   468][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  CLOCK
[   475][W][main.cpp:445] defaultsettings(): 0x0044,  OFF TIMER
[   482][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  VOLUME+
[   489][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  VOLUME-
[   495][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  << 30s
[   502][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  >> 30s
[   509][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  SLEEP
[   516][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  CANCEL
[   522][W][main.cpp:445] defaultsettings(): 0xFFFFFFFF,  -
[   529][W][main.cpp:449] defaultsettings(): 0x0001,  IR address
00:00:00 setup: ....  seek for WiFi networks
00:00:00 WiFI_info:   add SSID: mySSID
00:00:00 WiFI_info:   add SSID: Home_2487201_RPT_2
00:00:00 WiFI_info:   add SSID: Home_2487201
00:00:00 WiFI_info:   add SSID: Home_2487201_RPT
[  4033][W][STA.cpp:541] disconnect(): STA already disconnected.
00:00:00 WiFI_info:   Connecting WiFi...
00:00:00 WiFI_info:   mDNS name: MiniWebRadio
00:00:00 setup: ....  connected to Home_2487201_RPT_2, IP address is 192.168.1.73, 
Received Signal Strength -50 dB
00:00:00 ftpServer:   Buffers allocated: 4622 bytes
00:00:00 rtime_info:  Initializing SNTP
[... 156 lines cut ...]
00:00:00 setup: ....  volume steps: 100
00:00:00 setup: ....  volume after alarm: 10
00:00:00 setup: ....  last connected host: https://instrumental.stream.laut.fm/instrumental-orchestral-hits
00:00:00 setup: ....  connection timeout: 2500 ms
00:00:00 setup: ....  connection timeout SSL: 3500 ms
00:00:00 RESET_REASON:Software reset due to exception/panic
00:00:00 setup: ....  Ambient Light Sensor BH1750 found
12:02:54 ftpServer:   Ftp server waiting for connection on port 21
12:02:55 BT-Emitter:  KCX_BT_Emitter found 
12:02:55 BT-Emitter:  POWER ON 
12:02:55 BT-Emitter:  Version KCX_BT_RTX_V1.4
12:02:55 BT-Emitter:  Status -> Disconnected
12:02:55 BT-Emitter:  Volume -> 16
12:02:56 BT-Emitter:  OK+POWEROFF_MODE 
12:02:56 BT-Emitter:  Status -> Scan...
12:02:59 BT-Emitter:  Status -> Connected
12:03:00 BT-Emitter:  no response: last command was: AT+POWER_OFF
[ 13244][E][DLNAClient.cpp:231] readHttpHeader(): contentlength is not given
[ 13272][E][DLNAClient.cpp:375] getServerItems(): controlURL ?
[ 13296][E][DLNAClient.cpp:231] readHttpHeader(): contentlength is not given
[ 13304][E][DLNAClient.cpp:232] readHttpHeader(): content type not found
12:03:00 DLNA_server: [0] 192.168.1.4:6000  dlink-E247EB
12:03:00 DLNA_server: 1 media server found
12:03:06 BT-Emitter:  no response: last command was: AT+BT_MODE?
12:03:12 BT-Emitter:  no response: last command was: AT+VOL=16

schreibfaul1, 2025-01-24:
Hallo valera,
nachdem der erste Versuch nicht das gewünschte Ergebnis gebracht hat, hier der nächste:

valera39, 2025-01-24:
Hello Dear Wolle!
I tried to install the latest version with your changes, but I got an error #611
There are no errors in the file and the picture is not corrupted, it appears on the display. The reboot happens exactly when you open the web interface when playing this station

valera39, 2025-01-24:
Dear Wolle!
I solved the reboot problem. I replaced the ‘&’ with ‘and’ in the station name and in the picture name and everything works fine.
Please note that the presence of the ‘&’ sign is not allowed, as it causes a reboot. Maybe you can fix this bug.
Thank you for your attention to my problem.

schreibfaul1, 2025-01-24:
Very good! Some special characters cause problems. You cannot write file names with the & character to the SD card via the website. However, in operating systems such as Windows or Linux, the & is a valid character and can therefore be used as a file name.

closed 2025-01-25 by valera39