schreibfaul1/ESP32-MiniWebRadio #604: BBC HLS Radio deleted

RobinMosedale, 2025-01-21:
Radio 4 indicating 'Gone

https://[as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d48000.norewind.m3u8](https://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d48000.norewind.m3u8)

Responds Gone

R123 OK as is world service.

However R4, R4 extra, longwave and local radios reporting gone

RobinMosedale, 2025-01-21:
See https://gist.github.com/bpsib/67089b959e4fa898af69fea59ad74bc3
but need to clear flash and rebuild otherwise old url used

schreibfaul1, 2025-01-22:
Hi Robin,
it feels like the BBC changes the URLs every year.

http://lstn.lv/bbcradio.m3u8?station=bbc_radio_one&amp;bitrate=96000
is redirected to:
http://as-hls-ww-live.akamaized.net/pool_904/live/ww/bbc_1xtra/bbc_1xtra.isml/bbc_1xtra-audio%3d96000.norewind.m3u8 

% characters indicate URL encoding, %20 would be the space, for example. But that is not the case here.
In addition, I found a memory leak and a variable that was deleted too early.

But now it works:
http://lstn.lv/bbcradio.m3u8?station=bbc_radio_one&amp;bitrate=96000
http://lstn.lv/bbcradio.m3u8?station=bbc_1xtra&amp;bitrate=96000
... etc. are the new URLs.

Please reload the audioI2S library.

best regards.
Wolle