schreibfaul1/ESP32-MiniWebRadio #606: DACs funktionieren nicht

muhackl, 2025-01-22:
Hallo. Für das MiniWebRadio mit einem ESP32 S3 habe ich es mit folgenden beiden DACs versucht:
- PCM5102 
- MAX98357A

Leider kommt kein Ton aus den Lautsprechern bzw. beim PCM5102 auch nicht aus den Kopfhörern über den Klinkenanschluss.
Stromversorgung für den ESP32 S3 ist ein Netzteil mit 5,2 V und 1,8 A.
Ansonsten ist nur die SD-Karte angeschlossen. Kein Display, kein Bluetooth-Modul, kein IR-Sensor, kein BH1750.

Die Ausgabe in der Console sieht top aus - keine Fehlermeldungen, die auf eine Nichtfunktion des DACs schließen lässt. Die Weboberfläche ist erreichbar und es spielt offenbar ein Web-Radio-Stream. Nur hört man nichts. 

Lautsprecher sind beim PCM5102 z.B. über Lout und AGND angeschlossen. Die offenen Lötbrücken habe ich nach Anleitung geschlossen (außer die H3L, aber das sollte nichts ausmachen?)).
Beim MAX98357A gibt es nur einen Mono-Ausgang für einen Lautsprecher.

Ansonsten ist die Verkabelung mit Jumper-Kabel an die angelöteten Pins realisiert:

Beim PCM5102:
- DIN - GPIO9
- BCK - GPIO3
- LRCK - GPIO1
- SCK - GND
- VIN - 5V
- GND - GND

Beim MAX98357A:
- DIN - GPIO9 
- BCLK - GPIO3
- LRC - GPIO1
- SCK gibt es nicht
- VIN - 5V
- GND - GND
GAIN und SD habe ich nicht verbunden.

An was könnte es nun liegen, dass kein Ton aus dem Lautsprecher kommt? Auch kein Rauschen o.ä..
Vielen herzlichen Dank für eure Hilfe.

TubaFonse, 2025-01-22:
Servus Muhackl, 
ganz wos anderes - host scho moi g'schaut, ob "Mute" aktiv ist bzw. "Lautstärke auf Minimum"? Im Display wäre dann evtl. die Lautstärkeanzeige "Vol" in roter Farbe ... I bin do selber scho moi drübergstolpert ...

[image]

Vielleicht huifts, scheena Gruaß vom TubaFonse

schreibfaul1, 2025-01-22:
Vielleicht hast du keine 5V am DAC. Standardmäßig liegt am 5V Pin des ESP32-S3 Boards keine Spannung an.

Arne65, 2025-01-22:
Hast du die Brücken auf dem PCM gemacht ? 
SCK liegt auf GND ?

Siehe DOC Pcm5102

muhackl, 2025-01-23:
> Servus Muhackl, ganz wos anderes - host scho moi g'schaut, ob "Mute" aktiv ist bzw. "Lautstärke auf Minimum"? Im Display wäre dann evtl. die Lautstärkeanzeige "Vol" in roter Farbe ... I bin do selber scho moi drübergstolpert ...
> 
> [image]
> 
> Vielleicht huifts, scheena Gruaß vom TubaFonse

Servus Fonse,
do hob I scho drauf druckt. Ois grea. Sollt laffa. Aber trotzdem danke für'n Tip!

muhackl, 2025-01-23:
> Hast du die Brücken auf dem PCM gemacht ? SCK liegt auf GND ?
> 
> Siehe DOC Pcm5102

Siehe Beschreibung: 
- Brücken sind gemacht bis auf die H3L. Braucht es die superdringend?
- SCK liegt auf GND

muhackl, 2025-01-23:
> Vielleicht hast du keine 5V am DAC. Standardmäßig liegt am 5V Pin des ESP32-S3 Boards keine Spannung an.

Das ist ja über die Maßen unlogisch und inkonsequent. An den 3,3V-Pins liegen ja auch 3,3V an. 
Zudem stimmt dann das Anschlussschema in der Dokumentation nicht. Dort ist ja eingezeichnet, dass man den DAC an den 5V-Anschluss des ESP32 S3 anschließen soll. Muss ich jetzt wirklich einen eigenen Anschluss vom Netzteil an den 5V-Anschluss des DAC führen?

Arne65, 2025-01-23:
Moin Moin

eine extra Stromversorgung für das Modul benötigst du nicht.

Wenn die rote LED auf dem Modul leuchtet liegen erstmal die 5 V an und zumindest einer der beiden Spannungsregler auf dem Modul liefert 3,3 V. Auf dem Modul sind aber zwei verbaut.

Hast du alle Brücken so gemacht wie auf dem Schaltbild ? 
SCK liegt auf GND ?
in common.h den DAC PCM5102  also (1) ausgewählt ?
alles richtig verdrahtet  #define I2S_DOUT           9
                                      #define I2S_BCLK            3
                                      #define I2S_LRC             1?

Durch den ESP wird nicht geprüft ob der DAC angeschossen ist aus diesem Grund ist in deinem seriellen Monitor auch alles super Schick.

Wenn Du wirklich alles nach der Schaltplan so angeschlossen hast und wirklich beide 3,3 V ( siehe Schematic PCM5102a) inkl. Brücken gemacht hast liegt der verdacht nahe das dein DAC Modul nicht funktioniert.

Habe das mehrfach auch mit eigenen PCB Layouts gebaut .

skassel2024, 2025-01-23:
Diese Brücke beim ESP32-S3, sonst keine 5V an PIN "5Vin"!
[image]

muhackl, 2025-01-23:
> Moin Moin
>
> eine extra Stromversorgung für das Modul benötigst du nicht.
>
> Wenn die rote LED auf dem Modul leuchtet liegen erstmal die 5 V an und zumindest einer der beiden Spannungsregler auf dem [...]

Hi.
- ich habe alle Brücken verbunden wie auf dem Bild bzw. der Skizze - bis auf die H3L. Die ist nicht gebrückt.
- SCK liegt auf GND
- In der common.h ist für den DAC 1 ausgewählt. 
- Die Pins sind genau so definiert, wie sie dann auch angeschlossen sind.

Was ich nicht verstehe ist, was du mit "beide 3,3 V" gemeint hast. Die Skizze zeigt, dass ich den DAC an 5V anschließen soll. 
Bei dem Anschluss - bei genauerem Hinsehen und Blick in die Doku habe ich es jetzt auch gesehen - handelt es sich ja um einen Vin - also einen Eingang, der Strom "konsumiert" und nicht zur Verfügung stellt. Per Default zumindest. Im Post von @skassel2024 sieht man, dass man auf dem ESP32 S3 erst eine Brücke verbinden muss, dass daraus ein Ausgang wird. Das fehlt in der Doku oder ich habe es nicht gesehen. Wenn das so stimmt, dann brücke ich jetzt mal und sehe dann, ob der DAC Strom bekommt.

Arne65, 2025-01-23:
Hallöchen ,

will nochmal auf die 5V zurückkommen .

Die Brücke auf den ESP32-S3 Boards kenne ich nicht da ich nur originale ` Espressif Boards verwende ,da gibt es so eine Brücke nicht (zumindest nicht auf meinen) es geistern aber sehr viel Mutanten auf dem Markt rum und da mag es sowas durchaus geben .

Auf der Platine  des PSM5102 A befinden sich zwei ME6211C33M5G-N Spannungsregler 5V zu 3,3 V.

Schau dir das Schaltbild an dann siehst du es was ich meine 

[image]

Hoffe das du mittlerweile dem Teil etwas entlocken konntest .

Grüße Arne

skassel2024, 2025-01-24:
Einfach an Pin 5Vin messen, ob da auch 5V ankommen. Ansonsten den PCM direkt mit 5V vom Netzteil versorgen.
siehe auch Post in Arduino Forum, speziell die Diode bei Header 22 verhindert, dass da 5V anliegen. Daher die Brücke:
[Chinese ESP32-S3 5V pin WARNING](https://forum.arduino.cc/t/chinese-esp32-s3-5v-pin-warning/1192758)

muhackl, 2025-01-24:
@skassel2024 das war jetzt im Endeffekt auch die Lösung: Einfach den ESP mit 5V vom Netzteil versorgen am Vin-Pin und vom selben Pin gleich die 5V-Spannung für den DAC abgreifen. Läuft. 
Vorerst mit dem MAX98357A. Nächster Test mit dem PCM5102A.

skassel2024, 2025-01-24:
Auch beim Brücken sind beide USB-C noch nutzbar, da die 5V durch die Dioden D1 und D2 zu den Ports hin geblockt ist. Durch die Brücke wird nur D3 bei J1 PIN 21 überbrückt. Siehe dazu Schaltbild der China Version:

[image]