schreibfaul1/ESP32-MiniWebRadio #607: Web-Interface (1st Start): Station-No not synchron with display

TubaFonse, 2025-01-22:
Hallo Wolle, 
wenn das Web-Interface **zum ersten Mal gestartet** wird (Browser-Cache gelöscht), wird im oberen Feld (Station Number und Station Name) nicht der aktuell gespielte Sender angezeigt, sondern immer der erste aus der Senderliste. Das Sender-Logo ist jedoch korrekt das vom aktuell gespielten Sender. Nachdem der Sender einmal gewechselt wurde (einmal vor und zurück), stimmt Sendernummer und -Name in der oberen Zeile des Web-Interface. 
Könntest Du das bitte bei der Synchronisation (Start des Web-Interface) noch korrigieren? 

Danke und Gruß, 
Alfons. 

[image]

schreibfaul1, 2025-01-22:
Hi Alfons, 
das lag daran, dass der Index mit der Sendernummer übertragen wurde, bevor die Senderliste (von der SD Karte) fertig geladen war. Das habe ich korrigiert und es scheint zu funktionieren.
vG Wolle

TubaFonse, 2025-01-23:
Hallo Wolle, 
jetzt passts aber perfekt! Vielen Dank - Grüße, Alfons.

closed 2025-01-23 by TubaFonse