schreibfaul1/ESP32-MiniWebRadio #608: RGB HMI display

valera39, 2025-01-23:
Hello Dear Wolle!
I assume you are starting to design a new radio under the rgb hmi display. What kind of display do you recommend to buy and how to connect it to ESP32, using UART?
Will DMG80480C043_01W be suitable ?
If I am wrong please correct me.

valera39, 2025-01-24:
Well then when you get a chance, recommend which display to use

schreibfaul1, 2025-01-25:
It will be weeks before an HMI display works properly here. A display that is well suited would be the one from Elecrow https://www.elecrow.com/esp32-display-5-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html. But make sure that you don't buy an old version, the two marked transistors should be present. [image] But many other displays are not bad either.
The resolution must be 800x480px and preferably have a capacitive touch panel. These displays are available in 5 or 7 inch

valera39, 2025-01-25:
Unfortunately due to sanctions I can't purchase the display from this store. I will try to look for a similar one on Aliexpress

schreibfaul1, 2025-02-19:
Hello,
basically any display is suitable that has colour inputs (8 each for red, green and blue), control lines for horizontal and vertical synchronisation, a clock and a control input HSYNC, VSYNC, PCLK, DE). You would then have to connect the display to the ESP32-S3 with 20 wires.

It is easier if the ESP32-S3 is already on the display. Sometimes a DAC is on board, but it's still not perfect.

I have connected a DAC, a light sensor and an IR receiver externally to a Sunton display.

[image] 

[image] 

You will have to do something similar with the Elecrow display. The Elecrow display (5 or 7 inch) is well suited

valera39, 2025-02-19:
Thanks for the quick reply, very cool looking big display!
So I can safely buy this display you recommended? https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/url
Could you suggest a couple more specific links to suitable displays.
Also, would the N4R8 be too small?
Thank you, it's a pleasure to communicate with you!

valera39, 2025-02-19:
It's one like yours, right? https://wiki.makerfabs.com/Sunton_ESP32_S3_5_7_inch_IPS_with_Touch.html

schreibfaul1, 2025-02-19:
This is exactly the display I used.
https://wiki.makerfabs.com/Sunton_ESP32_S3_5_7_inch_IPS_with_Touch.html

Unlike an SPI display, it is not necessary to initialise the TFT controller or change the gamma values. 
But it is not possible to rotate or mirror the image.

valera39, 2025-02-19:
Great, so I'll be buying the same one
Thank you!

valera39, 2025-02-19:
How soon will there be a schematic of the connections to other components for this display?

schreibfaul1, 2025-02-19:
Please have a look at the V4 pre-release branch in the readme. There is a schematic diagram there. The use of the pins depends on the display.
With the ESP32-S3, 33 pins can be used (GPIO00-GPIO21 and GPIO38-GPIO48) 
Of these, 20 are needed for the display, leaving 22 pins.
3 pins for SD_MMC, 3 pins for the DAC, 2 pins for the touchpad and the brightness sensor. 1 for the display brightness, 1 for the IR receiver, 2 (TX0, RX0) for the serial terminal. 1 pin remains free.

valera39, 2025-02-20:
If we have a free output left, it can be assigned the value #define AMP_ENABLED
Would you add this?

schreibfaul1, 2025-02-20:
If you want to use this display, (almost) everything can remain as it is. The red GPIOs are for Flash and PSRAM and cannot be used. The two red marked GPIOs are not needed, there is no SD (but SD_MMC) and no TP_RST. The pins can be routed to the outside, but the track must be interrupted to prevent a malfunction.

AMP_ENABLED was used for ESP32 AI Thinker and OLIMEX boards, but I can activate it again.

[image]

valera39, 2025-02-20:
Yes, please add the AMP_ENABLED parameter again. I used it to control the power to the amplifier, turning it off when the radio goes to sleep. 
In the common.h file, for the Waveshare 7' Display, the parameter define TFT_BL -1, How then is the backlighting of the display controlled?
Waveshare 7' Display is more convenient for solderless connections, isn't it? But I'm more inclined to choose a display from Sunton, especially since you already have one working properly. What would you advise me ?

Ryszard-wpl, 2025-02-22:
If I will use Sunton display - remaining one free pin can be assign as MCLK for full I2S lines ? (in declaration you set it as -1 and it must be IMPORTANT!. Can I use full I2S lisnes or it will eat too much operational power of MCU? Thank you for information.

schreibfaul1, 2025-02-22:
DMA is required for I2S and also for the buffer that contains the image for the display. If you activate MCLK, the image is no longer synchronised.

valera39, 2025-02-22:
Dear Wolle!
So what can you say and advise?

schreibfaul1, 2025-02-22:
It's hard to say, I can't test all the displays that are out there.
I have a Waveshare 7 inch display with additional ICs for CAN bus, RS485 and I2C Portexpander. The additional ICs are not needed.
Then I have an Elecrow 5 inch display which is usable, it just shouldn't be version 1.1.
I like the Sunton 7 inch display best.
Unfortunately, none of the displays have all the pins we need externally on the existing connectors, so some remodelling is necessary.

It looks like all 800x480 displays have the TouchPad Controller GT911. It is important to pay attention to this with unknown displays.

If the display has a DAC on the board, it can be used (with restrictions) without modification.

valera39, 2025-02-22:
Thanks for the detailed response!
I'll probably order the Sunton 7 inch display.

schreibfaul1, 2025-03-06:
Hello @valera39,
I have added AMP_ENABLED. If you have a Sunton display you can use GPIO10 for it. This is intended for SD_CS, but is not used. However, you must ensure that the connection to the SD card is interrupted.

You can initially only use the KCX BT emitter as shown in the circuit. The receiver mode is currently not possible due to a lack of free GPIOs. But you can use it to connect to a BT speaker or BT headphones.

It may be possible to use the 16 RGB GPIOs twice as they are only used when something changes on the display. But that's something for later.

valera39, 2025-03-06:
I will be getting a Sunton display soon and will try it out. 
To increase the number of pins can apply GPIO expander PCF8574 or MCP23017. Is this possible?