schreibfaul1/ESP32-MiniWebRadio #610: Build error header file esp_lcd_panel_rgb.h not found for env [esp32]

rjshiffer, 2025-01-24:
Following error occurs during build of Env [esp32]:

```
In file included from src/common.h:47,
                 from src/main.cpp:1:
lib/tftLib/src/tft.h:12:10: fatal error: esp_lcd_panel_rgb.h:
         No such file or directory
```

Header file esp_lcd_panel_rgb.h is NOT found in:
```.platformio/packages/framework-arduinoespressif32-libs/esp32/include/esp_lcd/include```
                                                       
However, compilation succeeds for Env [esp32s3] as the header file IS found in:

```.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/esp_lcd/rgb/include```
                                                         
Note: The [esp32s3] success occurred before the latest release, as I now get
the same error described in issue #609.

Steps I have taken:
1. Full Clean
2. Remove and rebuild .platformio directory

schreibfaul1, 2025-01-24:
Hi @rjshiffer,
I'm in the process of adding an HMI display and didn't realise that the ESP32 doesn't come with the drivers for it. HMI displays require an ESP32-S3. I have corrected it.

rjshiffer, 2025-02-18:
Problem has re-occurred:
Version 3.7-rc1.g   - Feb 16/2025 

 In file included from src/common.h:48,
                 from src/main.cpp:1:
lib/tftLib/tft_rgb.h:15:10: fatal error: esp_lcd_panel_rgb.h: No such file or directory

schreibfaul1, 2025-02-19:
I copied the same error a second time :-((

Espressifs esp_lcd_panel_rgb is part of the libraries for RGB displays. This is not available for the ESP32.

closed 2025-04-05 by github-actions[bot]