schreibfaul1/ESP32-MiniWebRadio #611: Build error

valera39, 2025-01-24:
Hello Dear Wolle!
Build error of the current version. (Commit 7fd766a)

Compiling .pio\build\esp32s3\ZigbeeCarbonDioxideSensor.cpp.o
Compiling .pio\build\esp32s3\freertos\heap_idf.c.o
Archiving .pio\build\esp32s3\esp-idf\framework-arduinoespressif32\libframework-arduinoespressif32.a
\x91\xab\xa8誮\xac \xa4\xab\xa8\xad\xad\xa0\xef \xaa\xae\xac\xa0\xad\xa4\xad\xa0\xef \xe1\xe2ப\xa0.
*** [.pio\build\esp32s3\esp-idf\framework-arduinoespressif32\libframework-arduinoespressif32.a] Error 1
Full Clean and reinstalling Platformio did not help

valera39, 2025-01-25:
what about my problem?

schreibfaul1, 2025-01-25:
When compiler warnings like this come up, simply ignore them.

[image]

ETH.cpp:886:22: warning:[-Wdeprecated-declarations] or [-Wvolatile] or 
header is no longer used, and will be removed in future versions. [-Wcpp]

The warnings come from the Arduio or IDF framework

valera39, 2025-01-26:
So I have to wait for the fixes in Arduino IDF 5.3.2 to build successfully?
Now I can't compile the firmware on Arduino IDF 5.3.2 version

valera39, 2025-01-26:
With IDF version 5.4.0 everything compiled successfully

closed 2025-01-26 by valera39